SatDump/SatDump #3: Crashes while compositing images for M2 LRPT

jpjonte, 2021-03-16:
Hi,

I'm running into the following error while decoding an M2 LRPT baseband:

> satdump(83751,0x10fb31e00) malloc: *** error for object 0x7ffd1800d000: pointer being freed was not allocated
> satdump(83751,0x10fb31e00) malloc: *** set a breakpoint in malloc_error_break to debug
> [1]    83751 abort      ./satdump meteor_m2_lrpt baseband  products ~/Documents/Privat/sat/images/tmp

Full log: [satdump_output.txt](https://github.com/altillimity/SatDump/files/6149978/satdump_output.txt)

You can download the baseband [here](https://jpjonte.de/2021-03-14-18-09-m2-225000.raw).
Perhaps the recording is just too bad?

Any Idea what's going on here? I'm running SatDump on macOS 11.2.

arvedviehweger, 2021-03-16:
Hi,
I just ran your recording through SatDump on my computer that is running macOS 11.2.1.
I get a similar error in the end but it does produce products for me.

[03/16/21 - 15:33:42] (W) Processing data to level products
[03/16/21 - 15:33:42] (D) Parameters :
[03/16/21 - 15:33:42] (D)    - baseband_format : f32
[03/16/21 - 15:33:42] (D)    - samplerate : 225000
[03/16/21 - 15:33:42] (I) Using input frames /Users/arvedviehweger/temp/decode/mn22/meteor_m2_lrpt.cadu
[03/16/21 - 15:33:42] (I) Decoding to /Users/arvedviehweger/temp/decode/mn22/MSU-MR
[03/16/21 - 15:33:42] (I) Demultiplexing and deframing...
[03/16/21 - 15:33:50] (I) Progress 43.799999%
[03/16/21 - 15:33:55] (I) Writing images.... (Can take a while)
[03/16/21 - 15:33:55] (I) Channel 1...
[03/16/21 - 15:33:56] (I) Channel 2...
[03/16/21 - 15:33:56] (I) Channel 3...
[03/16/21 - 15:33:56] (I) Channel 4...
[03/16/21 - 15:33:56] (I) Channel 5...
[03/16/21 - 15:33:56] (I) Channel 6...
[03/16/21 - 15:33:56] (I) 221 Composite...
[03/16/21 - 15:33:58] (I) 321 Composite...
satdump(3475,0x119290e00) malloc: *** error for object 0x7fa8bc00b000: pointer being freed was not allocated
satdump(3475,0x119290e00) malloc: *** set a breakpoint in malloc_error_break to debug
zsh: abort      ./satdump meteor_m2_lrpt baseband  products  -samplerate 225000  f32

If I read the code correctly it produces all the outputs that it should be putting out.
When does it crash for you?

[m2.zip](https://github.com/altillimity/SatDump/files/6150297/m2.zip)

jpjonte, 2021-03-16:
It crashes at the same point as yours does (321 Composite).

The image for channel 3 is empty/0 bytes (makes sense, as not all channels are enabled). I guess it crashes because the composite is supposed to contain channel 3 and it doesn't expect the image to be empty.

arvedviehweger, 2021-03-16:
You are right, it seems that channel 3 is indeed empty. I added #4 to skip the 321 composite if there is no data in channel 3.
Although it doesn't fix the crash that is happening at the end just yet. That one will require more debugging.

Aang23, 2021-03-16:
Thanks! I tested with some of my files, and could not reproduce the issue even though I've had quite a few reports. Could one of you provide it? 
I will take a look at it in the meantime.

Aang23, 2021-03-17:
Thanks! Fixed it, turned out to be due to composites copying an array ending up in a double free.

closed 2021-03-17 by Aang23

jpjonte, 2021-03-17:
Awesome, thanks!

commit 3bca61b0 (2021-03-16): Merge pull request #4 from arvedviehweger/master

Fix another macOS CMake error and an attempt to fix #3
commit f4d8f991 (2022-09-18): Windows CI fixes #3