SatDump/SatDump #6: No error is produced when input file does not exist

shuyuan-liu, 2021-03-26:
## Description

If the command-line version of SatDump is given an input file path that does not exist, instead of giving an error, the program keeps printing `(I) Progress 100.000000%, Deframer : NOSYNC, Frames : 0` with seemingly no end.

The problem appears on both Release and Debug builds.

## Build info

- SatDump built from source, commit b8ec3e3
- gcc 10.2.0
- libdsp built from source, commit 783dc00
- libcorrect built from source, commit f5a28c7
- volk 2.4
- fftw 3.3.9

Aang23, 2021-05-10:
Indeed, definitely something I'd need to add a check for. Thanks!

Aang23, 2021-06-08:
Fixed in https://github.com/altillimity/SatDump/commit/9eb3c92bf5b35aede12c6d8368b58036901d4601

closed 2021-06-08 by Aang23

shuyuan-liu, 2021-06-09:
Thanks @Aang23!