SatDump/SatDump #11: Issues with Android build (macOS)

jpjonte, 2021-05-14:
Not sure if my dependencies are all setup correctly, but I'm having issues with the Android build on macOS.
I have the Android 11 SDK and NDK, installed through Android Studio 4.1.3.

If I run `./gradlew assembleDebug` I get the following error:

```
Task 'assembleDebug' not found in root project 'android'.
```

And running `./gradlew assemble` leads to:

```
BUILD SUCCESSFUL in 0s
```

Without anything being built.

Aang23, 2021-05-14:
This would occur if you do not have the ANDROID_SDK_ROOT Variable set (happened to me a few times). Perhaps make sure it is in the environment or in your local.properties?

jpjonte, 2021-05-14:
Yep, that was the issue. I copied local.properties from another project and it works fine now. Thanks!

closed 2021-05-14 by jpjonte