SatDump/SatDump #16: 'Live Processing' output?

K4KDR, 2021-06-07:
Just installed ver. 0.0.28 and tried the '**Live Processing**' option for the first time.  AMAZING work!  Thanks so much the the fun platform to decode weather sats.

I'm not sure that I am handling one aspect correctly.  The output of the 'Live Processing' session is a .CADU file.

That's great - I was able to run **`satdump-ui`** again and produce the Meteor-M2 images from the .CADU file.

But, is there a way to generate the end-result images in a single session?  Or, just as when using a .RAW recording from GQRX, is this still a 2-step process?   Thanks!!

[image]

janvgils, 2021-06-07:
That is looking good Scott.

I am still stuck in compiling the dependency `https://github.com/quiet/libcorrect.git` it will not compile on my debian testing system. I'll need to give this another try.

K4KDR, 2021-06-07:
Great!!  I had to manually update gcc, g++, & cmake to the latest versions on my Ubuntu 18.04 system.  The repository versions were not current enough

Aang23, 2021-06-09:
> Just installed ver. 0.0.28 and tried the '**Live Processing**' option for the first time. AMAZING work! Thanks so much the the fun platform to decode weather sats.
> 

Hi! Thanks!

> I'm not sure that I am handling one aspect correctly. The output of the 'Live Processing' session is a .CADU file.
> 
> That's great - I was able to run **`satdump-ui`** again and produce the Meteor-M2 images from the .CADU file.
> 

Yes, as of 0.0.28 the Live Processing feature was mostly designed to do part of the processing live. Eg, in the case of HRPT demodulation / decoding to save time on post-processing and ease tracking. Hence why the output of most live-processing pipelines in a .cadu.
Except for a few satellites (GOES HRIT, GVAR, FengYun-2 S-VISSR...), it's in general how it works as not everything is worth running live / that easy to get working as well live. (A demodulator / decoder is "streaming" by nature. Takes data in, sends data out. Imagery decoders aggregate it all and then outputs it)

> But, is there a way to generate the end-result images in a single session? Or, just as when using a .RAW recording from GQRX, is this still a 2-step process? Thanks!!
> 

In 0.0.28, no. But I added a "finish processing" feature in https://github.com/altillimity/SatDump/commit/812f8cde1b87160d8f00a8d60d39722d572b627e, which will optionally (A checkbox has been added) launch the rest of the offline processing when done. This should make it all a 1-step process as you had to press stop already anyway :-)
[image]

This will not be made a default feature as I also need to cover other use cases such as HRPT / etc where there can be overlapping passes, and waiting for some processing may be undesirable. (Though, the SDR is released first which means you could use a second SatDump instance for this purpose)

> That is looking good Scott.
> 
> I am still stuck in compiling the dependency `https://github.com/quiet/libcorrect.git` it will not compile on my debian testing system and also a conflict with gnuradio-dev . I'll need to give this another try.

Out of curiosity? Which package was conflicting with gnuradio-dev? I would guess it was VOLK. In general you can stick with older Volk versions as SatDump will include missing headers itself. Though, when doing so you must make sure to do what is explained in the README, as otherwise Viterbi decoding could go wrong... (I have a PR for this in Volk, hopefully that will get merged soon enough)

> Great!! I had to manually update gcc, g++, & cmake to the latest versions on my Ubuntu 18.04 system. The repository versions were not current enough

Yes... Ubuntu 18.04 is starting to get a bit old, Linux went a long way since then! Unfortunately I can't really get around that issue, best option is either to do as you did and upgrade, or I could consider producing .AppImage files or something like flatpak packages. Though as of now it hasn't been too much of an issue as far as I know.

K4KDR, 2021-06-09:
Thank you so much for the explanations!  Two steps are fine, but I'll check out the update in 812f8cd !

Regarding AppImage, that is a great system.  I run new versions of GQRX now using the AppImage download and it runs perfectly.

Aang23, 2021-06-09:
You're welcome!
Now I'd rather advise you use 0.0.29 though. (or to be honest, feel free to use master directly. In general I don't break things and if I do fix, I usually fix it really quickly)

I think I will consider building AppImage packages starting from the next release, I won't recommend it as a default installation method on Linux (building from source / install is still preferable), but it will fill the gaps for users that are having more trouble than that or just wish to run it without going through the trouble.

(To clarify : Building from source will always result in better performances and as an AppImage is self-contained would not allow to customize pipelines / resources such as TLE files as intended without some trickery.)

K4KDR, 2021-06-09:
Wonderful!  Yes, I got the notification ref. 0.0.29 & will definitely download that.

Thanks for everything!

K4KDR, 2021-06-09:
... and just did a quick over-the-air test (Airspy > Meteor-M2 LRPT) using v0.0.29 w/ the "Finish Processing" box check - worked perfectly!

Great work & many thanks.

Aang23, 2021-06-10:
Awesome, thanks for checking!
You're welcome!

I would guess this issue can be closed now?

janvgils, 2021-06-10:
_Out of curiosity? Which package was conflicting with gnuradio-dev? I would guess it was VOLK. In general you can stick with older Volk versions as SatDump will include missing headers itself. Though, when doing so you must make sure to do what is explained in the README, as otherwise Viterbi decoding could go wrong... (I have a PR for this in Volk, hopefully that will get merged soon enough)_

Be leafing the system with libvolk2-dev I could keep using gnuradio-dev and build SatDump.

Now I am puzzled with what input options are available, there only seems to be a file input method and no SDR devices, are there dependencies towards the supported SDR devices that need to be available?

Also the life processing isn't available and complains about it being rewritten.

So I guess in need to do some further investigation on what is and can be expected after a master branch build.

K4KDR, 2021-06-10:
Hi Jan!

Regarding the 'Live Processing' option, BEFORE doing any of the build steps did you edit **`CMakeLists.txt`** to enable that feature?

[image]

... and if you have already built the app once, please start over by deleting the **build** folder:

`rm -rf build`

... and with 'Live Processing' enabled in **`CMakeLists.txt`**, do all those steps again.

Hope that helps!

K4KDR, 2021-06-10:
**_>>> I would guess this issue can be closed now?_**

You answered my questions - thank you so much.  Please see above ref. Jan's follow-up.

janvgils, 2021-06-10:
Hi Scott,

Thanks for putting me in the right direction, building as we speak.

All the best

Aang23, 2021-06-10:
> Be leafing the system with libvolk2-dev I could keep using gnuradio-dev and build SatDump.
> 

That is entirely fine, libvolk2-dev is actually preferred above libvolk1-dev, but legacy systems usually do not have it. I guess I should make this clear in the REDME then.

> Now I am puzzled with what input options are available, there only seems to be a file input method and no SDR devices, are there dependencies towards the supported SDR devices that need to be [...]

You need to replace the `cmake -DCMAKE_BUILD_TYPE=Release ..` command by `cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_LIVE=ON ..` to enable it. Once again something I should add in the READMe I guess :-)

> Hi Jan!
>
> Regarding the 'Live Processing' option, BEFORE doing any of the build steps did you edit **`CMakeLists.txt`** to enable that feature?
>
> ... and if you have already built the app once, please [...]

That was definitely in the right direction, but a bit overcomplicated. No need to delete the contents of build after doing such a modified (though, the correct way to do it is what I described above), just running the cmake command again will update everything required and build as expected.

> You answered my questions - thank you so much. Please see above ref. Jan's follow-up.

You're welcome! Yes, of course I will before closing this :-)

Aang23, 2021-08-10:
As this has been idle for a while, I'm going to close this issue.
Feel free to ask for a reopen is required though!

closed 2021-08-10 by Aang23

miguelvrr, 2021-11-10:
Hi, I have already the newest build for win with the live processing aparently working. Satdump is able to open my airspy but the elektro L TLM signal is non present. I  can't decode anything and the signal stays on 0.0 Any idea how to configure the software to receive this?? All help is welcome! Thx