SatDump/SatDump #19: Segfault while decoding FY3B ERM

jpjonte, 2021-06-15:
Hi,

I just encountered a segfault while decoding FY3B. This is the shell output:

> [06/15/21 - 17:25:00] (I) Using input frames /Users/bud/Documents/private/sat/recordings/2021-06-15-17-04-f3b/fengyun3_ab_ahrpt_live.cadu
> [06/15/21 - 17:25:00] (I) Decoding to /Users/bud/Documents/private/sat/recordings/2021-06-15-17-04-f3b/ERM
> [06/15/21 - 17:25:00] (I) Demultiplexing and deframing...
> [06/15/21 - 17:25:00] (I) Progress 0.000000%
> [06/15/21 - 17:25:00] (I) VCID 12 Frames         : 240
> [06/15/21 - 17:25:00] (I) CCSDS Frames           : 146
> [06/15/21 - 17:25:00] (I) ERM Lines              : 4
> [06/15/21 - 17:25:00] (I) Writing images.... (Can take a while)
> [06/15/21 - 17:25:00] (I) Channel 1...
> [1]    43540 segmentation fault  satdump-ui

FYI this was recorded with a Nooelec Smart v4 at 3 MSps so I didn't expect this to result in anything anyway.

CADUs are here:

[fengyun3_ab_ahrpt_live.cadu.zip](https://github.com/altillimity/SatDump/files/6656830/fengyun3_ab_ahrpt_live.cadu.zip)

Either way thanks for making this great software!

Aang23, 2021-06-16:
Hi!
Thanks! This is an issue I knew about but never was able to reproduce on my end. Until now, as your file (finally!) crashes on my end as well.

> Either way thanks for making this great software!

You're welcome :-)

closed 2021-06-16 by Aang23

jpjonte, 2021-06-17:
Thanks, works fine now!

commit 2f7f079b (2021-06-16): Fixup ERM Decoder crashing. Fixes #19