SatDump/SatDump #20: MacOS build error

benb0jangles, 2021-06-25:
I'm getting this error during build:

Command: 
`make -j4`

Error:
```
[ 34%] Building CXX object src-core/CMakeFiles/satdump_core.dir/init.cpp.o
/Users/benji/satdump/src-core/init.cpp:5:10: fatal error: 'filesystem' file not
      found
#include <filesystem>
         ^~~~~~~~~~~~
1 error generated.
make[2]: *** [src-core/CMakeFiles/satdump_core.dir/init.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [src-core/CMakeFiles/satdump_core.dir/all] Error 2
make: *** [all] Error 2
```

Please help?

Using:
MacOS 10.12.6 Sierra

benb0jangles, 2021-06-25:
Problem found with xcode 9.2:

> Xcode 11 Beta now includes <filesystem>. Unlike the other answers indicating beta support in Xcode 10, Apple has mentioned this in the release notes.
> 
> Also mentioned in the release notes, is this is only supported by iOS 13, macOS 10.15, watchOS 6, and tvOS 13. You will only be able to use std::filesystem for projects targeting these versions or later.

[](https://stackoverflow.com/questions/42633477/macos-clang-c17-filesystem-header-not-found)

Aang23, 2021-06-28:
Hi!
From what I see, this is because your compiler does not support some required C++17 features...

I would guess you could either update your compiler, or perhaps I should consider providing a fallback (boost libraries) in case it is missing.

Though, maybe if you cannot update your compiler, you could use GCC from brew. Might result in degraded performances though...

benb0jangles, 2021-06-28:
> Hi!
> From what I see, this is because your compiler does not support some required C++17 features...
>
> I would guess you could either update your compiler, or perhaps I should consider providing a [...]

I think it is Xcode 9.2 on my iMac Sierra - it does not have 'filesystem', I will try to compile on my mocabook pro catalina and see if compiling is working, then message back

Aang23, 2021-08-10:
Out of curiosity, did you ever figure this one out @benb0jangles?

benb0jangles, 2021-08-15:
> Out of curiosity, did you ever figure this one out @benb0jangles?

I am yet to test on catalina macbook, but I will over the next couple of weeks. Please bear with :)

benb0jangles, 2021-10-13:
Install by:
Make sure you have git, homebrew, xcode installed

then # macOS: Install dependencies
brew install cmake 
brew install volk 
brew install jpeg 
brew install libpng 
brew install glew 
brew install glfw 
brew install nng

# Build and install libcorrect
git clone https://github.com/quiet/libcorrect.git
cd libcorrect
mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=Release ..              
make -j4
sudo make install
cd ../..
rm -rf libcorrect

# macOS ONLY: build and install libfftw3
# if you install fftw via brew, cmake won't be able to find it
wget http://www.fftw.org/fftw-3.3.9.tar.gz
tar xf fftw-3.3.9.tar.gz
rm fftw-3.3.9.tar.gz
cd fftw-3.3.9
mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTS=false -DENABLE_FLOAT=true ..
make
sudo make install
cd ../..
rm -rf fftw-3.3.9

# Finally, SatDump
git clone https://github.com/altillimity/satdump.git
cd satdump
mkdir build && cd build
# If you want to build Live-processing (required for the ingestor), add -DBUILD_LIVE=ON to the command
# If you do not want to build the GUI Version, add -DNO_GUI=ON to the command
# If you want to disable some SDRs, you can add -DENABLE_SDR_AIRSPY=OFF or similar
cmake -DCMAKE_BUILD_TYPE=Release ..                             # MacOS
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr .. # Linux
make -j4
ln -s ../pipelines . # Symlink pipelines so it can run
ln -s ../resources . # Symlink pipelines so it can run
ln -s ../Ro* . # Symlink fonts for the GUI version so it can run

# Run (if you want!)
./satdump-ui