SatDump/SatDump #30: 6M sample rate with an Airspy produces no FFT data in live processing LRPT

shuyuan-liu, 2021-08-10:
Hi! I've recently got an Airspy Mini and tried to receive LRPT with it. I found that if I set the sample rate to 6000000 in live processing LRPT, after pressing Start all graphs are static and no signal / data seems to be coming through from the Airspy. Admittedly 6M is a bit too high for a signal less than 200kHz wide and the more sensible 3M works fine, but still this is a bug that needs to be looked at (at some point).

[image]

[image]

Aang23, 2021-08-10:
Indeed. That should not be happening. Do you have the satdump.log file on hand? I would want to see what the resampler what set to.

shuyuan-liu, 2021-08-12:
Using 6M,
```
[08/12/21 - 10:41:19] (D) Starting livedemod...
[08/12/21 - 10:41:19] (I) meteor_m2_lrpt
[08/12/21 - 10:41:19] (I) Starting live pipeline METEOR M2 LRPT
[08/12/21 - 10:41:19] (I) Module qpsk_demod
[08/12/21 - 10:41:19] (D) Parameters :
[08/12/21 - 10:41:19] (D)    - agc_rate : 0.1
[08/12/21 - 10:41:19] (D)    - baseband_format : f32
[08/12/21 - 10:41:19] (D)    - buffer_size : 8192
[08/12/21 - 10:41:19] (D)    - costas_bw : 0.004
[08/12/21 - 10:41:19] (D)    - dc_block : 0
[08/12/21 - 10:41:19] (D)    - rrc_alpha : 0.6
[08/12/21 - 10:41:19] (D)    - rrc_taps : 361
[08/12/21 - 10:41:19] (D)    - samplerate : 6000000
[08/12/21 - 10:41:19] (D)    - symbolrate : 72000
[08/12/21 - 10:41:19] (I) Module meteor_lrpt_decoder
[08/12/21 - 10:41:19] (D) Parameters :
[08/12/21 - 10:41:19] (D)    - baseband_format : f32
[08/12/21 - 10:41:19] (D)    - dc_block : 0
[08/12/21 - 10:41:19] (D)    - diff_decode : 0
[08/12/21 - 10:41:19] (D)    - samplerate : 6000000
[08/12/21 - 10:41:19] (T) Volk has the spiral kernel, using it!
[08/12/21 - 10:41:19] (D) Starting SDR...
[08/12/21 - 10:41:19] (D) Device parameters airspy:
[08/12/21 - 10:41:19] (D)    - bias : 1
[08/12/21 - 10:41:19] (D)    - gain : 10
[08/12/21 - 10:41:19] (I) Opened Airspy device!
[08/12/21 - 10:41:19] (I) Samplerate 6000000.000000
[08/12/21 - 10:41:19] (I) Frequency 137100000.000000
[08/12/21 - 10:41:19] (D) Input SPS : 83.333336
[08/12/21 - 10:41:19] (D) Resample : 1
[08/12/21 - 10:41:19] (D) Samplerate : 210000.000000
[08/12/21 - 10:41:19] (D) Dec factor : 28.571428
[08/12/21 - 10:41:19] (D) Final SPS : 2.916667
[08/12/21 - 10:41:19] (I) Start processing...
[08/12/21 - 10:41:19] (I) Using input baseband 
[08/12/21 - 10:41:19] (I) Demodulating to /mnt/Data/Pictures/Satellite/meteor_m2_lrpt.soft
[08/12/21 - 10:41:19] (I) Buffer size : 237568
[08/12/21 - 10:41:19] (I) Start processing...
[08/12/21 - 10:41:19] (I) Using input symbols 
[08/12/21 - 10:41:19] (I) Decoding to /mnt/Data/Pictures/Satellite/meteor_m2_lrpt.cadu
...
```

Using 3M,
```
[08/12/21 - 10:39:56] (D) Starting livedemod...
[08/12/21 - 10:39:56] (I) meteor_m2_lrpt
[08/12/21 - 10:39:56] (I) Starting live pipeline METEOR M2 LRPT
[08/12/21 - 10:39:56] (I) Module qpsk_demod
[08/12/21 - 10:39:56] (D) Parameters :
[08/12/21 - 10:39:56] (D)    - agc_rate : 0.1
[08/12/21 - 10:39:56] (D)    - baseband_format : f32
[08/12/21 - 10:39:56] (D)    - buffer_size : 8192
[08/12/21 - 10:39:56] (D)    - costas_bw : 0.004
[08/12/21 - 10:39:56] (D)    - dc_block : 0
[08/12/21 - 10:39:56] (D)    - rrc_alpha : 0.6
[08/12/21 - 10:39:56] (D)    - rrc_taps : 361
[08/12/21 - 10:39:56] (D)    - samplerate : 3000000
[08/12/21 - 10:39:56] (D)    - symbolrate : 72000
[08/12/21 - 10:39:56] (I) Module meteor_lrpt_decoder
[08/12/21 - 10:39:56] (D) Parameters :
[08/12/21 - 10:39:56] (D)    - baseband_format : f32
[08/12/21 - 10:39:56] (D)    - dc_block : 0
[08/12/21 - 10:39:56] (D)    - diff_decode : 0
[08/12/21 - 10:39:56] (D)    - samplerate : 3000000
[08/12/21 - 10:39:56] (T) Volk has the spiral kernel, using it!
[08/12/21 - 10:39:56] (D) Starting SDR...
[08/12/21 - 10:39:56] (D) Device parameters airspy:
[... 41 lines cut ...]
[08/12/21 - 10:40:16] (D)    - samplerate : 1000000
[08/12/21 - 10:40:16] (T) Volk has the spiral kernel, using it!
[08/12/21 - 10:40:16] (D) Starting SDR...
[08/12/21 - 10:40:16] (D) Device parameters airspy:
[08/12/21 - 10:40:16] (D)    - bias : 0
[08/12/21 - 10:40:16] (D)    - gain : 10
[08/12/21 - 10:40:16] (I) Opened Airspy device!
[08/12/21 - 10:40:16] (I) Samplerate 1000000.000000
[08/12/21 - 10:40:16] (I) Frequency 137100000.000000
[08/12/21 - 10:40:16] (D) Input SPS : 13.888889
[08/12/21 - 10:40:16] (D) Resample : 1
[08/12/21 - 10:40:16] (D) Samplerate : 210000.000000
[08/12/21 - 10:40:16] (D) Dec factor : 4.761905
[08/12/21 - 10:40:16] (D) Final SPS : 2.916667
[08/12/21 - 10:40:16] (I) Start processing...
[08/12/21 - 10:40:16] (I) Start processing...
[08/12/21 - 10:40:16] (I) Using input baseband 
[08/12/21 - 10:40:16] (I) Demodulating to /mnt/Data/Pictures/Satellite/meteor_m2_lrpt.soft
[08/12/21 - 10:40:16] (I) Buffer size : 40960
[08/12/21 - 10:40:16] (I) Using input symbols 
[08/12/21 - 10:40:16] (I) Decoding to /mnt/Data/Pictures/Satellite/meteor_m2_lrpt.cadu
...
```

Only 6M doesn't work.

shuyuan-liu, 2021-08-13:
The tests above were made with dae96ec3442e6645ab7371eaf28faea9447da575. I've now updated to 2cca8179c64c1a5c9379bc9aa9651cb461b7208c and the problem persists. No difference in terminal output between the two versions.

I don't know if it's related or even normal, but at higher sample rates the constellation diagrams seem to be moving more slowly. Since this is live decoding, shouldn't symbols be coming at the same rate no matter the sample rate?

Aang23, 2021-08-13:
> The tests above were made with [dae96ec](https://github.com/altillimity/SatDump/commit/dae96ec3442e6645ab7371eaf28faea9447da575). I've now updated to [2cca817](https://github.com/altillimity/SatDump/commit/2cca8179c64c1a5c9379bc9aa9651cb461b7208c) and the problem persists. No difference in terminal output between the two versions.
> 
> I don't know if it's related or even normal, but at higher sample rates the constellation diagrams seem to be moving more slowly. Since this is live decoding, shouldn't symbols be coming at the same rate no matter the sample rate?

I just fixed it in https://github.com/altillimity/SatDump/commit/fac23f91bd218bdc44dd6931a2544134e1bf78a4.
Even though it is the same samplerate, resampling is done on large buffers and yield less updates per second on very high ratios. It's something that could be fixed but at the cost of some performance... Don't think it's that necessary.

closed 2021-08-13 by Aang23