SatDump/SatDump #32: airspyhf errors on 'make' stage v0.0.37

K4KDR, 2021-08-13:
The 'make' is failing for me w/ v0.0.37; I'm sure it's an error or omission on my part, but I'm not finding what that is.   

I do **not** own an AirSpy-HF, but "think" I have all the listed dependencies installed.

Long list of errors are:

```
[ 93%] Building CXX object libsdr/CMakeFiles/sdr.dir/sdr/hackrf.cpp.o
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp: In static member function ‘static int SDRAirspyHF::_rx_callback(airspyhf_transfer_t*)’:
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp:11:92: warning: ‘void* memcpy(void*, const void*, size_t)’ copying an object of non-trivial type ‘struct std::complex<float>’ from an array of ‘struct airspyhf_complex_float_t’ [-Wclass-memaccess]
     std::memcpy(stream->writeBuf, t->samples, t->sample_count * sizeof(std::complex<float>));
                                                                                            ^
In file included from /usr/local/include/volk/volk_complex.h:22,
                 from /usr/local/include/volk/volk_typedefs.h:30,
                 from /usr/local/include/volk/volk.h:29,
                 from /home/k4kdr/SatDump-0.0.37/libsdr/../src-core/common/dsp/buffer.h:4,
                 from /home/k4kdr/SatDump-0.0.37/libsdr/sdr/sdr.h:3,
                 from /home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.h:4,
                 from /home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp:1:
/usr/include/c++/8/complex:1072:12: note: ‘struct std::complex<float>’ declared here
     struct complex<float>
            ^~~~~~~~~~~~~~
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp: In member function ‘virtual void SDRAirspyHF::start()’:
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp:67:5: error: ‘airspyhf_set_hf_agc’ was not declared in this scope
     airspyhf_set_hf_agc(dev, agc_mode != 0);
     ^~~~~~~~~~~~~~~~~~~
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp:67:5: note: suggested alternative: ‘airspyhf_set_freq’
     airspyhf_set_hf_agc(dev, agc_mode != 0);
     ^~~~~~~~~~~~~~~~~~~
     airspyhf_set_freq
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp:68:5: error: ‘airspyhf_set_hf_agc_threshold’ was not declared in this scope
     airspyhf_set_hf_agc_threshold(dev, agc_mode - 1);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp:68:5: note: suggested alternative: ‘airspyhf_set_calibration’
     airspyhf_set_hf_agc_threshold(dev, agc_mode - 1);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
     airspyhf_set_calibration
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp:70:5: error: ‘airspyhf_set_hf_att’ was not declared in this scope
     airspyhf_set_hf_att(dev, attenuation / 6.0f);
     ^~~~~~~~~~~~~~~~~~~
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp:70:5: note: suggested alternative: ‘airspyhf_set_freq’
     airspyhf_set_hf_att(dev, attenuation / 6.0f);
     ^~~~~~~~~~~~~~~~~~~
     airspyhf_set_freq
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp:71:5: error: ‘airspyhf_set_hf_lna’ was not declared in this scope
     airspyhf_set_hf_lna(dev, hf_lna);
     ^~~~~~~~~~~~~~~~~~~
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp:71:5: note: suggested alternative: ‘airspyhf_set_freq’
     airspyhf_set_hf_lna(dev, hf_lna);
     ^~~~~~~~~~~~~~~~~~~
     airspyhf_set_freq
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp: In member function ‘virtual void SDRAirspyHF::drawUI()’:
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp:104:9: error: ‘airspyhf_set_hf_att’ was not declared in this scope
         airspyhf_set_hf_att(dev, attenuation / 6.0f);
         ^~~~~~~~~~~~~~~~~~~
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp:104:9: note: suggested alternative: ‘airspyhf_set_freq’
         airspyhf_set_hf_att(dev, attenuation / 6.0f);
         ^~~~~~~~~~~~~~~~~~~
         airspyhf_set_freq
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp:111:9: error: ‘airspyhf_set_hf_agc’ was not declared in this scope
         airspyhf_set_hf_agc(dev, agc_mode != 0);
         ^~~~~~~~~~~~~~~~~~~
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp:111:9: note: suggested alternative: ‘airspyhf_set_freq’
         airspyhf_set_hf_agc(dev, agc_mode != 0);
         ^~~~~~~~~~~~~~~~~~~
         airspyhf_set_freq
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp:112:9: error: ‘airspyhf_set_hf_agc_threshold’ was not declared in this scope
         airspyhf_set_hf_agc_threshold(dev, agc_mode - 1);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp:112:9: note: suggested alternative: ‘airspyhf_set_calibration’
         airspyhf_set_hf_agc_threshold(dev, agc_mode - 1);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         airspyhf_set_calibration
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp:117:9: error: ‘airspyhf_set_hf_lna’ was not declared in this scope
         airspyhf_set_hf_lna(dev, hf_lna);
         ^~~~~~~~~~~~~~~~~~~
/home/k4kdr/SatDump-0.0.37/libsdr/sdr/airspyhf.cpp:117:9: note: suggested alternative: ‘airspyhf_set_freq’
         airspyhf_set_hf_lna(dev, hf_lna);
         ^~~~~~~~~~~~~~~~~~~
         airspyhf_set_freq
libsdr/CMakeFiles/sdr.dir/build.make:89: recipe for target 'libsdr/CMakeFiles/sdr.dir/sdr/airspyhf.cpp.o' failed
make[2]: *** [libsdr/CMakeFiles/sdr.dir/sdr/airspyhf.cpp.o] Error 1
```

THANK YOU!

Aang23, 2021-08-13:
I see 2 solutions : Either disable the support with -DENABLE_SDR_AIRSPYHF=OFF in the cmake arguments, or installing libairspyhf-dev which I would guess is missing.
You're welcome!

I should write code to disable SDRs that do not have a library detected.

K4KDR, 2021-08-13:
Thanks so much!

`libairspyhf-dev` **was** installed, but apparently that was not enough.

If anyone else runs into this issue, the CMAKE command that got me going was:

`cmake -DCMAKE_BUILD_TYPE=Release -DENABLE_SDR_AIRSPYHF=OFF -DCMAKE_INSTALL_PREFIX=/usr ..`

Aang23, 2021-08-13:
Weird for sure!

I have pushed a commit that will only build SDR Support for a given SDR if the right library is found. That should avoid this kind of issues.
Perhaps your libairspyhf was too old?

K4KDR, 2021-08-13:
>>> Perhaps your libairspyhf was too old?

... don't know - it's what is currently available in Ubuntu 18.04:

```
k4kdr@3010i5:~/SatDump-0.0.37/build$ apt-cache policy libairspyhf-dev
libairspyhf-dev:
  Installed: 1.0-1
  Candidate: 1.0-1
  Version table:
 *** 1.0-1 500
        500 http://us.archive.ubuntu.com/ubuntu bionic/universe amd64 Packages
        100 /var/lib/dpkg/status
```

... but no matter, the switch to disable certain SDRs and now the update you've added to auto-check the installed SDR support takes care of it just fine.  Many thanks as always!!

Aang23, 2021-08-13:
Yes, then that is the issue. I am on 1.6.8 here and 1.0 is very outdated...
Good that the rest worked though!

You're welcome! Gonna close this now, I think that's it.

closed 2021-08-13 by Aang23

benb0jangles, 2021-10-15:
Similar error experienced here too compiling SatDump v0.0.37:

```
/home/user2/satdump/libsdr/sdr/airspyhf.cpp: In member function ‘virtual void SDRAirspyHF::start()’:
/home/user2/satdump/libsdr/sdr/airspyhf.cpp:67:5: error: ‘airspyhf_set_hf_agc’ was not declared in this scope; did you mean ‘airspyhf_set_freq’?
   67 |     airspyhf_set_hf_agc(dev, agc_mode != 0);
      |     ^~~~~~~~~~~~~~~~~~~
      |     airspyhf_set_freq
/home/user2/satdump/libsdr/sdr/airspyhf.cpp:68:5: error: ‘airspyhf_set_hf_agc_threshold’ was not declared in this scope
   68 |     airspyhf_set_hf_agc_threshold(dev, agc_mode - 1);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user2/satdump/libsdr/sdr/airspyhf.cpp:70:5: error: ‘airspyhf_set_hf_att’ was not declared in this scope; did you mean ‘airspyhf_set_freq’?
   70 |     airspyhf_set_hf_att(dev, attenuation / 6.0f);
      |     ^~~~~~~~~~~~~~~~~~~
      |     airspyhf_set_freq
/home/user2/satdump/libsdr/sdr/airspyhf.cpp:71:5: error: ‘airspyhf_set_hf_lna’ was not declared in this scope; did you mean ‘airspyhf_set_freq’?
   71 |     airspyhf_set_hf_lna(dev, hf_lna);
      |     ^~~~~~~~~~~~~~~~~~~
      |     airspyhf_set_freq
/home/user2/satdump/libsdr/sdr/airspyhf.cpp: In member function ‘virtual void SDRAirspyHF::drawUI()’:
/home/user2/satdump/libsdr/sdr/airspyhf.cpp:104:9: error: ‘airspyhf_set_hf_att’ was not declared in this scope; did you mean ‘airspyhf_set_freq’?
  104 |         airspyhf_set_hf_att(dev, attenuation / 6.0f);
      |         ^~~~~~~~~~~~~~~~~~~
      |         airspyhf_set_freq
/home/user2/satdump/libsdr/sdr/airspyhf.cpp:111:9: error: ‘airspyhf_set_hf_agc’ was not declared in this scope; did you mean ‘airspyhf_set_freq’?
  111 |         airspyhf_set_hf_agc(dev, agc_mode != 0);
      |         ^~~~~~~~~~~~~~~~~~~
      |         airspyhf_set_freq
/home/user2/satdump/libsdr/sdr/airspyhf.cpp:112:9: error: ‘airspyhf_set_hf_agc_threshold’ was not declared in this scope
  112 |         airspyhf_set_hf_agc_threshold(dev, agc_mode - 1);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user2/satdump/libsdr/sdr/airspyhf.cpp:117:9: error: ‘airspyhf_set_hf_lna’ was not declared in this scope; did you mean ‘airspyhf_set_freq’?
  117 |         airspyhf_set_hf_lna(dev, hf_lna);
      |         ^~~~~~~~~~~~~~~~~~~
      |         airspyhf_set_freq
libsdr/CMakeFiles/sdr.dir/build.make:89: recipe for target 'libsdr/CMakeFiles/sdr.dir/sdr/airspyhf.cpp.o' failed
make[2]: *** [libsdr/CMakeFiles/sdr.dir/sdr/airspyhf.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:266: recipe for target 'libsdr/CMakeFiles/sdr.dir/all' failed
make[1]: *** [libsdr/CMakeFiles/sdr.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 92%] Linking CXX executable ../satdump
[ 92%] Built target satdump
Makefile:135: recipe for target 'all' failed
make: *** [all] Error 2
```
using K4KDR suggestion above to disable airspyhf worked for me
https://github.com/altillimity/SatDump/issues/32#issuecomment-898605641

System:
Arm/Aarch64 Jetson Nano
Distributor ID: Ubuntu
Description: Ubuntu 18.04.6 LTS
Release: 18.04
Codename: bionic
JetPack 4.6
upgraded cmake version to 3.21.3
upgraded to gcc-10 & g++-10

So now also working on Nvidia Jetson Nano, but I need to test how well performance is during Live Decoding of Meteor, maybe tomorrow :)
[image]