SatDump/SatDump #33: Support for the sdrplay ?

funnymdzz, 2021-08-21:
I have a sdrplay rsp1 but I can't use it with SatDump's live processing Is there any plan to support sdrplay devices

funnymdzz, 2021-08-21:
I use rsp_tcp with SatDump but failed the fluctuates greatly

bclswl0827, 2021-08-21:
Hi, you can try with [SDRplay/RSPTCPServer](https://github.com/SDRplay/RSPTCPServer), a `rtl_tcp` compatible program for SDRPlay.

Aang23, 2021-08-21:
Hi, I do plan to support SDRPlay devices once I have a way to test things. I have reached out to SDRPlay about it, so hopefully that will be possible shortly.

Can't really support devices I can't properly test my code with.

In the meantime, you will unfortunately have to record baseband or use another way of feeding the data such as what @bclswl0827 mentioned.

funnymdzz, 2021-08-22:
> Hi, you can try with [SDRplay/RSPTCPServer](https://github.com/SDRplay/RSPTCPServer), a `rtl_tcp` compatible program for SDRPlay.

I have tested rsp_tcp but it failed I can open the live processing but cannot decode and the signal cannot be locked
gnuradio with xrit-rx can works well and decodes successfully

funnymdzz, 2021-08-22:
> Hi, I do plan to support SDRPlay devices once I have a way to test things. I have reached out to SDRPlay about it, so hopefully that will be possible shortly.
>
> Can't really support devices I can't [...]

Thanks for your reply and I will try more to make it work

Aang23, 2021-08-23:
> > Hi, you can try with [SDRplay/RSPTCPServer](https://github.com/SDRplay/RSPTCPServer), a `rtl_tcp` compatible program for SDRPlay.
> 
> I have tested rsp_tcp but it failed I can open the live processing but cannot decode and the signal cannot be locked
> gnuradio with xrit-rx can works well and decodes successfully for the gk-2a satellite

Are you sure the samplerate you set is valid? I doubt rsp_tcp does any resampling so you need to make sure the samplerate you input is supported by the RSP1A.

funnymdzz, 2021-08-23:
> Are you sure the samplerate you set is valid? I doubt rsp_tcp does any resampling so you need to make sure the samplerate you input is supported by the RSP1A.

I tried to set the samplerate of rsp_tcp and satdump to be the same, but it didn’t work. I think sdrplay has a 14bit ADC and rtl_sdr is 8bit. For compatibility, rsp_tcp will convert 14bit to 8bit.

bclswl0827, 2021-08-31:
Hi, `rsp_tcp` works fine with SatDump, I'm decoding FY-2H these days and I did a brief test (180cm dish + SDRplay 1A)
[image]

bclswl0827, 2021-08-31:
[image]
Successfully received FY-2H full disk image via RSP1A, hope this helps!

funnymdzz, 2021-08-31:
> Hi, `rsp_tcp` seems to work fine with SatDump, I'm decoding FY-2H these days and I did a brief test (180cm dish + SDRplay 1A)
> [image]
I used a sdrplay rsp1 not a rsp1a it has a DC spike I used the windows version of satdump it does not have rsp_tcp control

bclswl0827, 2021-08-31:
> I tried to set the samplerate of rsp_tcp and satdump to be the same, but it didn’t work. I think sdrplay has a 14bit ADC and rtl_sdr is 8bit. For compatibility, rsp_tcp will convert 14bit to 8bit.

My inference is that `rsp_tcp` throws redundant data in order to be compatible with `rtl_tcp`, however, this caused performance loss. In my opinion, SatDump should try to interface with SoapySDR rather than just supporting the SDRplay series, which means that more SDRs will be able to use SatDump, instead of manually recording baseband and using offline demodulation.

funnymdzz, 2021-08-31:
> My inference is that `rsp_tcp` throws redundant data in order to be compatible with `rtl_tcp`, however, this caused performance loss. In my opinion, SatDump should try to interface with SoapySDR [...]

agree with you

Aang23, 2021-09-05:
> My inference is that `rsp_tcp` throws redundant data in order to be compatible with `rtl_tcp`, however, this caused performance loss. In my opinion, SatDump should try to interface with SoapySDR [...]

I have considered SoapySDR and it has been my go-to in the past, but from my experience and others', it does come with pretty huge caveats and flaws meaning it's in the end... A lot more pain than implementing proper support for each SDR.
Then, some modules are straight up broken in SoapySDR, such as the SDRPlay one.

That's why I have not implemented Soapy support and don't plan on supporting it that much. I could add it as an option, but don't want to spend time debugging every single issue that's not from my side.

In the end, I am making my own SDR abstraction layer, but you pointed out it requires doing the support work manually for each device, which of course is a lot easier when I have access to the hardware for testing... It has downside but in this instance and in a software like SatDump the control this gives over how I'll deal with the SDR and such comes extremely handy in the end.

And well, to be honest on another point about SoapySDR, some required features are not properly abstracted so making it work still requires quite a lot of per-device specifics. If I'm going to need to do that kind of work anyway, implementing the entire thing myself with a lower-level API becomes really worth it.