SatDump/SatDump #37: Decimation?

K4KDR, 2021-09-04:
I continue to enjoy SatDump greatly!

Is there any possibility of adding 'decimation' to the "Live processing" screen?

It would be wonderful to be able to narrow the passband closer to the expected signal's bandwidth.    A high sample rate, then decimated, can often produce better SNR also.

Thank you!

-Scott,  K4KDR

Aang23, 2021-09-05:
Hi!
Decimation is actually already done internally, in the demodulator. Taking LRPT, the signal will automatically be resampled at 3x the symbolrate (216kHz) maximum. That is done to avoid insane processing time or other issues when using overkill samplerates such as 10Mhz or so which would usually cause the clock recovery to mess up at such oversampling.

You can hence feel free to use any samplerate (RTL-SDR Maximum 2.4 / 2.56M though! Over that samples will get dropped).

However, as that is done in the demodulator it does not show in the FFT Window, which gets fed the full spectrum.

K4KDR, 2021-09-05:
Thank you!!  Very good to know.  I was just going by the visual in the FFT & wondered if my SNR could be improved by some decimation.  Great to know that is taken care of!

closed 2022-01-05 by Aang23