SatDump/SatDump #41: linux- last make -j4 error

thowilwin0, 2021-09-14:
every other command up until this point was working fine, but when i got to the last `make -j4` I got the `make: *** No targets specified and no makefile found.  Stop.` error, I am not good at using the UNIX command line, or coding at any level so i might be completely be glossing over something. The command that I put in previously was this `cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE` if that helps.
BTW I'm using linux Mint witch is a ubontu based system. here is some more info.
`NAME="Linux Mint"
VERSION="20 (Ulyana)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 20"
VERSION_ID="20"
HOME_URL="https://www.linuxmint.com/"
`
thank you for your time

Aang23, 2021-09-14:
Hi, at first sight is appears you simply did not run a valid cmake command. It should be `cmake -DCMAKE_BUILD_TYPE=Release ..` at least. Maybe you copied the end of the line even though it's actually longer in the README?
make -j4 returning that result definitely confirms CMake failed to properly generate build files.

thowilwin0, 2021-09-14:
thank you for responding to my inquiry, i deeply appreciate it.  i tried your advice and i got a different error from the cmake command: `cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr ..` i also tried `cmake -DCMAKE_BUILD_TYPE=Release .. `
here is the return message from both of them:
_INSTALL_PREFIX=/usr .. 
-- Checking for module 'volk'
--   No package 'volk' found
CMake Error at /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:463 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:643 (_pkg_check_modules_internal)
  src-core/CMakeLists.txt:89 (pkg_check_modules)

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
JPEG_LIBRARY
    linked by target "satdump_core" in directory /home/thoma/satdump/src-core

-- Configuring incomplete, errors occurred!
See also "/home/thoma/satdump/build/CMakeFiles/CMakeOutput.log".
See also "/home/thoma/satdump/build/CMakeFiles/CMakeError.log".

closed 2022-01-27 by ZbychuButItWasTaken