SatDump/SatDump #42: SatDump on M1 - Volk without M1 support - Cant finish installing

juanibuqt, 2021-09-21:
Im having the following issue installing SatDump.
I do all the steps, until the following>

cmake -DCMAKE_BUILD_TYPE=Release ..

I get>

nox@Onion build % cmake -DCMAKE_BUILD_TYPE=Release ..
-- Checking for module 'volk'
--   No package 'volk' found
CMake Error at /opt/local/share/cmake-3.21/Modules/FindPkgConfig.cmake:562 (message):
  A required package was not found
Call Stack (most recent call first):
  /opt/local/share/cmake-3.21/Modules/FindPkgConfig.cmake:784 (_pkg_check_modules_internal)
  src-core/CMakeLists.txt:89 (pkg_check_modules)

Im using macOS Big Sur 11.6 running on MBPRo M1

As far as I can track, there is an issue of Volk supporting M1, im following this: 
https://github.com/gnuradio/volk/issues/428

Maybe there is any patch or workaround to get rid of this, anyway its useful as an issue/report for people installing SatDump on M1

benb0jangles, 2021-10-14:
tried either:
sudo port install volk
or
brew install volk
?

Aang23, 2021-10-17:
Hi! Sorry for the very late answer, I've been quite busy...
First of all did you install Volk as @benb0jangles pointed out? You may have to build it from source on M1.

Others have had success running Volk on M1 Macs for a while, and someone has also been using SatDump successfully, so I'd highly suspect some weirdness with installing it in the first place.

I'd still think the current guide applies though, but I of course do not have the hardware to test myself. (Though I do have a Big Sur x86_64 VM, and I can confirm everything works there)

juanibuqt, 2021-12-02:
Hi there! sorry for my late reply, I tested @benb0jangles solution but none of them worked. 
What worked and started was: brew install --build-from-source volk

Here is the full log

nox@Onion-2 satdump % sudo port install volk
Password:
Error: Current platform "darwin 21" does not match expected platform "darwin 20"
Error: If you upgraded your OS, please follow the migration instructions: https://trac.macports.org/wiki/Migration
OS platform mismatch
    while executing
"mportinit ui_options global_options global_variations"
Error: /opt/local/bin/port: Failed to initialize MacPorts, OS platform mismatch
nox@Onion-2 satdump % brew install volk

Running `brew update --preinstall`...
==> Auto-updated Homebrew!
Updated 2 taps (homebrew/core and homebrew/cask).
==> New Formulae
fastfec                                  goplus                                   mt32emu                                  pocsuite3                                sevenzip
==> Updated Formulae
Updated 153 formulae.
==> New Casks
appflowy
==> Updated Casks
Updated 95 casks.

Error: volk: no bottle available!
You can try to install from source with:
  brew install --build-from-source volk
Please note building from source is unsupported. You will encounter build
failures with some formulae. If you experience any issues please create pull
requests instead of asking for help on Homebrew's GitHub, Twitter or any other
official channels.
nox@Onion-2 satdump %   brew install --build-from-source volk

==> Downloading https://files.pythonhosted.org/packages/5c/db/2d2d88b924aa4674a080aae83b59ea19d593250bfe5ed789947c21736785/Mako-1.1.4.tar.gz
Already downloaded: /Users/nox/Library/Caches/Homebrew/downloads/1192562b4dca939fb599c936f921b2bc31279f6271b1c48262601ac079a76bf7--Mako-1.1.4.tar.gz
==> Cloning https://github.com/gnuradio/volk.git
Updating /Users/nox/Library/Caches/Homebrew/volk--git
==> Checking out tag v2.5.0
HEAD is now at 237a6fc Release 2.5.0
HEAD is now at 237a6fc Release 2.5.0
Entering 'cpu_features'
/Users/nox/Library/Caches/Homebrew/volk--git/cpu_features
==> python3 -m venv --system-site-packages /opt/homebrew/Cellar/volk/2.5.0/libexec/venv
==> /opt/homebrew/Cellar/volk/2.5.0/libexec/venv/bin/pip install -v --no-deps --no-binary :all: --ignore-installed /private/tmp/volk--Mako-20211202-85037-8oxt69/Mako-1.1.4
==> cmake . -DPYTHON_EXECUTABLE=/opt/homebrew/Cellar/volk/2.5.0/libexec/venv/bin/python -DENABLE_TESTING=OFF
==> make install
Last 15 lines from /Users/nox/Library/Logs/Homebrew/volk/04.make:
cd /tmp/volk-20211202-85037-fbaqja/lib && /opt/homebrew/Library/Homebrew/shims/mac/super/clang -DHAVE_DLFCN_H -DHAVE_FENV_H -DHAVE_POSIX_MEMALIGN -D_GLIBCXX_USE_CXX11_ABI=1 -I/opt/homebrew/include/orc-0.4 -I/tmp/volk-20211202-85037-fbaqja/include -I/tmp/volk-20211202-85037-fbaqja/kernels -I/tmp/volk-20211202-85037-fbaqja/lib -I/tmp/volk-20211202-85037-fbaqja/cpu_features/include -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -fvisibility=hidden -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk -fPIC -std=gnu11 -MD -MT lib/CMakeFiles/volk_obj.dir/volk_malloc.c.o -MF CMakeFiles/volk_obj.dir/volk_malloc.c.o.d -o CMakeFiles/volk_obj.dir/volk_malloc.c.o -c /tmp/volk-20211202-85037-fbaqja/lib/volk_malloc.c
[ 82%] Building C object lib/CMakeFiles/volk_obj.dir/volk.c.o
cd /tmp/volk-20211202-85037-fbaqja/lib && /opt/homebrew/Library/Homebrew/shims/mac/super/clang -DHAVE_DLFCN_H -DHAVE_FENV_H -DHAVE_POSIX_MEMALIGN -D_GLIBCXX_USE_CXX11_ABI=1 -DLV_MACHINE_GENERIC_ORC -DLV_MACHINE_NEONV8 -DLV_MACHINE_NEON_ORC -I/opt/homebrew/include/orc-0.4 -I/tmp/volk-20211202-85037-fbaqja/include -I/tmp/volk-20211202-85037-fbaqja/kernels -I/tmp/volk-20211202-85037-fbaqja/lib -I/tmp/volk-20211202-85037-fbaqja/cpu_features/include -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -fvisibility=hidden -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk -fPIC -std=gnu11 -MD -MT lib/CMakeFiles/volk_obj.dir/volk.c.o -MF CMakeFiles/volk_obj.dir/volk.c.o.d -o CMakeFiles/volk_obj.dir/volk.c.o -c /tmp/volk-20211202-85037-fbaqja/lib/volk.c
[ 83%] Building C object lib/CMakeFiles/volk_obj.dir/volk_cpu.c.o
cd /tmp/volk-20211202-85037-fbaqja/lib && /opt/homebrew/Library/Homebrew/shims/mac/super/clang -DHAVE_DLFCN_H -DHAVE_FENV_H -DHAVE_POSIX_MEMALIGN -D_GLIBCXX_USE_CXX11_ABI=1 -DVOLK_CPU_FEATURES=1 -I/opt/homebrew/include/orc-0.4 -I/tmp/volk-20211202-85037-fbaqja/include -I/tmp/volk-20211202-85037-fbaqja/kernels -I/tmp/volk-20211202-85037-fbaqja/lib -I/tmp/volk-20211202-85037-fbaqja/cpu_features/include -Wall -Werror=incompatible-pointer-types -Werror=pointer-sign -fvisibility=hidden -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk -fPIC -std=gnu11 -MD -MT lib/CMakeFiles/volk_obj.dir/volk_cpu.c.o -MF CMakeFiles/volk_obj.dir/volk_cpu.c.o.d -o CMakeFiles/volk_obj.dir/volk_cpu.c.o -c /tmp/volk-20211202-85037-fbaqja/lib/volk_cpu.c
[ 84%] Building C object lib/CMakeFiles/volk_obj.dir/volk_machines.c.o
cd /tmp/volk-20211202-85037-fbaqja/lib && /opt/homebrew/Library/Homebrew/shims/mac/super/clang -DHAVE_DLFCN_H -DHAVE_FENV_H -DHAVE_POSIX_MEMALIGN -D_GLIBCXX_USE_CXX11_ABI=1 -DLV_MACHINE_GENERIC_ORC -DLV_MACHINE_NEONV8 -DLV_MACHINE_NEON_ORC -I/opt/homebrew/include/orc-0.4 -I/tmp/volk-20211202-85037-fbaqja/include -I/tmp/volk-20211202-85037-fbaqja/kernels -I/tmp/volk-20211202-85037-fbaqja/lib -I/tmp/volk-20211202-85037-fbaqja/cpu_features/inc [... cut ...] ity=hidden -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk -fPIC -std=gnu11 -funsafe-math-optimizations -funsafe-math-optimizations -MD -MT lib/CMakeFiles/volk_obj.dir/volk_machine_neonv8.c.o -MF CMakeFiles/volk_obj.dir/volk_machine_neonv8.c.o.d -o CMakeFiles/volk_obj.dir/volk_machine_neonv8.c.o -c /tmp/volk-20211202-85037-fbaqja/lib/volk_machine_neonv8.c
[ 87%] Built target volk_obj
make: *** [all] Error 2

READ THIS: https://docs.brew.sh/Troubleshooting

AlexandreRouma, 2021-12-02:
You need to delete the version of cpu_features that was cloned along with volk and replace it with a fork that supports M1 (search the pull requests on the cpu_features github)

Aang23, 2022-09-16:
I will close this for now as this does not appear to be an issue anymore.

closed 2022-09-16 by Aang23