SatDump/SatDump #43: fftw

benb0jangles, 2021-09-23:
on satdump page:

```
# macOS ONLY: build and install libfftw3
# if you install fftw via brew, cmake won't be able to find it
wget http://www.fftw.org/fftw-3.3.9.tar.gz
tar xf fftw-3.3.9.tar.gz
rm fftw-3.3.9.tar.gz
cd fftw-3.3.9
mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTS=false -DENABLE_FLOAT=true ..
make
sudo make install
cd ../..
rm -rf fftw-3.3.9
```
question:
how to then create symlink of /Users/usrname/fftw-3.3.9/build/libfftw3f.3.dylib
to /usr/local/opt/fftw/lib/libfftw3f.3.dylib

???

AlexandreRouma, 2021-09-23:
`sudo make install` takes care of it

benb0jangles, 2021-09-23:
> `sudo make install` takes care of it

does not make symlink of /Users/usrname/fftw-3.3.9/build/libfftw3f.3.dylib
to /usr/local/opt/fftw/lib/libfftw3f.3.dylib

Must symlink using terminal command somehow