SatDump/SatDump #44: No R-S lock without App Restart

K4KDR, 2021-09-28:
This is very repeatable and I have not been able to find a cause.

System is Ubuntu 18.04, SatDump v0.0.37, "Live processing" option for Meteor-M2 137 MHz LRPT, Airspy-R2 @ 2.5M Samplerate

When satdump-ui GUI interface is launched, all appears well.  At AOS, the signal on the FFT grows & grows as elevation increases.  Viterbi turns green & the 4-dot constellations look normal.  However, no matter how strong the signal or how long you wait, Reed-Solomon will stay red & there is no decode.

[image]

However, if you un-check "Finish processing" and click "Stop", the click "Start" to restart the live capture, Reed-Solomon immediately turns green & images decode & are processed normally.

[image]

Any suggestions on how I might be able to stop using this work-around much appreciated!

Aang23, 2021-10-17:
@K4KDR, do you think you could record a baseband once and share it? (Using the same settings / etc). I will then try to reproduce the issue on my side.

There was a bug like that back before I released anything, but it is supposed to be fixed...

@marianzb this appears to be a RTL-TCP issue. Mind opening another issue about this?

K4KDR, 2021-10-17:
I assume you mean a recording when it is NOT locked?  

Yes, will do!    Just to be clear, is it the .CADU file you are looking for?

Thanks!

Aang23, 2021-10-17:
No, I meant a baseband (maybe preferrably using SatDump's recorder in f32 mode, as it uses the same DSP chain). That way I can feed it back into the demod / decode and see where something fails / etc.

K4KDR, 2021-10-17:
I spoke too soon - I can record a 32-bit .raw I/Q with GQRX, but I have no idea how to "us(e) SatDump's recorder in f32 mode".

Any guidance appreciated!

Aang23, 2021-10-17:
Oh right, I did not release the recorder in 0.0.37... So either use the current source on Git or GQRX, should do too since I doubt the issue lies in the SDR part.

AlexandreRouma, 2021-10-17:
Without looking at the code I bet the incoming data bytes get desyncronized and that causes the Q value from one sample to be read with the I value of the next sample. This won't have much effect on the constellation but will mess up the symbols

Aang23, 2021-10-18:
Wait, @K4KDR, could you also see if enabling DC Blocking helps? Some R2 units (mostly older ones it seems) sometimes had some I/Q offset that could effectively cause this kind of issues. If it's the case it wouldn't be too hard to fix at least.

K4KDR, 2021-10-18:
Very glad to enable the 'DC Block' option.  I normally do not since the Airspy doesn't have that issue.   I'll have that checked on future passes just to see if this issue does or does not continue.  Thanks!

K4KDR, 2021-10-18:
Well, it's only one attempt but on tonight's M2 pass (w/ 'DC Block' enabled), SatDump achieved RS-Lock without having to be restarted.  I will continue to use the 'DC-Block' option to see if that continues to give me a lock when SNR is good.

[image]

[image]

K4KDR, 2021-11-06:
[v0.0.38 / master branch / build: 01a8276, built from source]

Running 'Live Processing' with Airspy-R2 @ 2.5M Sample Rate:

On the **first live Meteor-M2 LRPT pass** with this build,  the **Reed-Solomon** indicator locked green without having to stop & re-start the GUI.   However, since this was an intermittent issue, I will continue to monitor the situation & verify that it locks each time without the need for a stop & re-start.   Thanks very much for the updates!

[image]

marianzb, 2021-12-20:
it looks my problem with RS locking was caused by some jitter on rtl-tcp server. Recently I have reinstalled Raspberry Pi there and now live decoding works perfectly all indicators green. :-) Also updated latest version, thanks for updating this project!

Aang23, 2022-01-05:
RTL-TCP has this kind of issue pretty often... So not too surprised it was that.

I guess this can be closed now!

closed 2022-01-05 by Aang23