SatDump/SatDump #57: AirSpy-R2 10M ok, 2.5M error

K4KDR, 2021-10-30:
After downloading the updated v0.0.38 code, I can now run 'Live Processing' with my Airspy-R2 at the 10M sample rate setting, but if I enter 2.5M, I get the following:

-------------------------------------------------------------------------------------------
```
[10/30/21 - 00:06:23] (E) No settings file found! Using defaults.
[10/30/21 - 00:06:40] (D) Starting livedemod...
[10/30/21 - 00:06:42] (D) Opening file dialog
[10/30/21 - 00:06:50] (D) Dir /home/k4kdr/SatDump-0.0.38/OUTPUT
[10/30/21 - 00:06:51] (D) Starting livedemod...
[10/30/21 - 00:06:51] (I) meteor_m2_lrpt
[10/30/21 - 00:06:51] (I) Starting live pipeline METEOR M2 LRPT
[10/30/21 - 00:06:51] (I) Module qpsk_demod
[10/30/21 - 00:06:51] (D) Parameters :
[10/30/21 - 00:06:51] (D)    - agc_rate : 0.1
[10/30/21 - 00:06:51] (D)    - baseband_format : f32
[10/30/21 - 00:06:51] (D)    - buffer_size : 8192
[10/30/21 - 00:06:51] (D)    - costas_bw : 0.004
[10/30/21 - 00:06:51] (D)    - dc_block : 0
[10/30/21 - 00:06:51] (D)    - rrc_alpha : 0.6
[10/30/21 - 00:06:51] (D)    - rrc_taps : 361
[10/30/21 - 00:06:51] (D)    - samplerate : 2500000
[10/30/21 - 00:06:51] (D)    - symbolrate : 72000
[10/30/21 - 00:06:51] (I) Module meteor_lrpt_decoder
[10/30/21 - 00:06:51] (D) Parameters :
[10/30/21 - 00:06:51] (D)    - baseband_format : f32
[10/30/21 - 00:06:51] (D)    - dc_block : 0
[10/30/21 - 00:06:51] (D)    - diff_decode : 0
[10/30/21 - 00:06:51] (D)    - samplerate : 2500000
[10/30/21 - 00:06:51] (T) Volk has the spiral kernel, using it!
[10/30/21 - 00:06:51] (D) Starting SDR...
[10/30/21 - 00:06:51] (D) Device parameters airspy:
[10/30/21 - 00:06:51] (I) Opened Airspy device!
[10/30/21 - 00:06:51] (I) Samplerate 2500000.000000
[10/30/21 - 00:06:51] (I) Frequency 137100000.000000
[10/30/21 - 00:06:51] (D) Input SPS : 34.722221
[10/30/21 - 00:06:51] (D) Resample : 1
[10/30/21 - 00:06:51] (D) Samplerate : 210000.000000
[10/30/21 - 00:06:51] (D) Dec factor : 11.904762
[10/30/21 - 00:06:51] (D) Final SPS : 2.916667
corrupted size vs. prev_size
Aborted (core dumped)
```
-------------------------------------------------------------------------------------------

... as you can see at the top of that dump, I deleted my **settings.json** file to make sure that nothing saved in there was causing this issue.

FYI, the image files that resulted from the 'Live' 10M capture appeared normal, but it would be great to be able to run with the 2.5M sample rate to have a more useful FFT display and to reduce the system load on the computer.

Many thanks!     -Scott,  K4KDR

Aang23, 2021-10-30:
That is very weird, 2.5M works just fine here.

Can you try using the latest code on Github first to be sure?

K4KDR, 2021-10-30:
Many thanks for the recommendation!

The result was the same... 10M sample rate runs normally; 2.5M fails (output below)

```
[10/30/21 - 17:50:50] (D) Starting livedemod...
[10/30/21 - 17:50:50] (I) Generated folder name : /home/k4kdr/SatDump/output/2021-10-30_17-50_meteor_m2_lrpt_137Mhz
[10/30/21 - 17:50:50] (I) meteor_m2_lrpt
[10/30/21 - 17:50:50] (I) Starting live pipeline METEOR M2 LRPT
[10/30/21 - 17:50:50] (I) Module qpsk_demod
[10/30/21 - 17:50:50] (D) Parameters :
[10/30/21 - 17:50:50] (D)    - agc_rate : 0.1
[10/30/21 - 17:50:50] (D)    - baseband_format : f32
[10/30/21 - 17:50:50] (D)    - buffer_size : 8192
[10/30/21 - 17:50:50] (D)    - costas_bw : 0.004
[10/30/21 - 17:50:50] (D)    - dc_block : 0
[10/30/21 - 17:50:50] (D)    - rrc_alpha : 0.6
[10/30/21 - 17:50:50] (D)    - rrc_taps : 361
[10/30/21 - 17:50:50] (D)    - samplerate : 2500000
[10/30/21 - 17:50:50] (D)    - symbolrate : 72000
[10/30/21 - 17:50:50] (I) Module meteor_lrpt_decoder
[10/30/21 - 17:50:50] (D) Parameters :
[10/30/21 - 17:50:50] (D)    - baseband_format : f32
[10/30/21 - 17:50:50] (D)    - dc_block : 0
[10/30/21 - 17:50:50] (D)    - diff_decode : 0
[10/30/21 - 17:50:50] (D)    - samplerate : 2500000
[10/30/21 - 17:50:50] (T) Volk has the spiral kernel, using it!
[10/30/21 - 17:50:50] (D) Starting SDR...
[10/30/21 - 17:50:50] (D) Device parameters airspy:
[10/30/21 - 17:50:50] (I) Opened Airspy device!
[10/30/21 - 17:50:50] (I) Samplerate 2500000.000000
[10/30/21 - 17:50:50] (I) Frequency 137100000.000000
[10/30/21 - 17:50:50] (D) Input SPS : 34.722221
[10/30/21 - 17:50:50] (D) Resample : 1
[10/30/21 - 17:50:50] (D) Samplerate : 210000.000000
[10/30/21 - 17:50:50] (D) Dec factor : 11.904762
[10/30/21 - 17:50:50] (D) Final SPS : 2.916667
[10/30/21 - 17:50:50] (I) Start processing...
[10/30/21 - 17:50:50] (I) Start processing...
[10/30/21 - 17:50:50] (I) Using input baseband 
[10/30/21 - 17:50:50] (I) Demodulating to /home/k4kdr/SatDump/output/2021-10-30_17-50_meteor_m2_lrpt_137Mhz/meteor_m2_lrpt.soft
[10/30/21 - 17:50:50] (I) Buffer size : 98304
[10/30/21 - 17:50:50] (I) Using input symbols 
[10/30/21 - 17:50:50] (I) Decoding to /home/k4kdr/SatDump/output/2021-10-30_17-50_meteor_m2_lrpt_137Mhz/meteor_m2_lrpt.cadu
Segmentation fault (core dumped)

k4kdr@3010i5:~/SatDump/build$ git branch
* master
```

Aang23, 2021-10-30:
What are you attempting to do exactly? Because I cannot reproduce it anymore here...

K4KDR, 2021-10-30:
Running satdump-ui, 'Live Processing', Meteor M2 LRPT, 2.5M Sample Rate

[image]

KentuckyFriedData, 2021-10-30:
I can reproduce this as well and get

[image]

i have even included logs below

[gdb-debug.txt](https://github.com/altillimity/SatDump/files/7447635/gdb-debug.txt)
[valgrind-debug.txt](https://github.com/altillimity/SatDump/files/7447636/valgrind-debug.txt)

Aang23, 2021-11-01:
That is the exact bug I fixed recently. I will try to reproduce on Ubuntu 18.4 I guess... Can't on my machine.

Aang23, 2021-11-06:
Actually, https://github.com/altillimity/SatDump/commit/01a8276de8c18cf24f455919abb44a9d7dc2ec10 may also have fixes this issue. Would anyone be able to confirm?

K4KDR, 2021-11-06:
Will update now & check on tomorrow's pass.

Many thanks!

K4KDR, 2021-11-06:
Sorry - don't need a satellite to test whether it crashes w/ the 2.5M Sample Rate or not.

Works great!   Thanks so much for the update.  Can run 'Live' w/ the 2.5M rate again.

[image]

K4KDR, 2021-11-06:
[v0.0.38 / master branch / build: 01a8276, built from source]

Running 'Live Processing' with Airspy-R2 @ 2.5M Sample Rate:

satdump-ui 'Live Processing' ran properly with the **2.5M Sample Rate** - thank you!!

[image]

Aang23, 2021-11-28:
I guess I can close this now!

closed 2021-11-28 by Aang23