SatDump/SatDump #58: v0.0.38 no values for 'SNR' or 'Peak SNR' displayed

K4KDR, 2021-10-31:
Since the upgrade to v0.0.38, there are no values for 'SNR' or 'Peak SNR' in the 'Signal' box (QPSK Demodulator window)

[System:   Ubuntu 18.04, Satdump build master / g463f45f]

[image]

Thanks!   -Scott,  K4KDR

Aang23, 2021-11-01:
That is weird... Though that constellation looks very wrong as well. Would you be able to record a baseband? Is that local noise causing this?

K4KDR, 2021-11-01:
Glad to!  Just looked at the 'Baseband Recorder' tab in v0.0.38 for the first time... seems that it runs with both 10M and 2.5M sample rates, so only the 'Live Processing' tab crashes with 2.5M.

Would you like the Recording with the 'i8', 'i16', or 'f32' option.  2.5M or 10M sample rate?

Thanks!

K4KDR, 2021-11-01:
Please disregard my last question - tonight's Meteor-M2 pass is here, so I'll record using 2.5M in the i16 format.

K4KDR, 2021-11-02:
Ok, baseband recorded but at 3.8G, I would need overnight to upload it so please let me know if that might be of any use to you.   As usual for M2, the signal looked great tonight & the baseband recorder worked perfectly at the 2.5M sample rate:

[image]

... and the resulting image (just a bit cropped off the top & bottom) turned out just about perfect.  While the 'Live Processing' option is fun, I certainly have no problem recording & then using the 'Offline Processing' tab to generate the images!

[image]

Aang23, 2021-11-05:
> Glad to! Just looked at the 'Baseband Recorder' tab in v0.0.38 for the first time... seems that it runs with both 10M and 2.5M sample rates, so only the 'Live Processing' tab crashes with 2.5M.
> 
> Would you like the Recording with the 'i8', 'i16', or 'f32' option. 2.5M or 10M sample rate?
> 
> Thanks!

i8 would suffice I'd say.

Good that it worked from baseband, but it pretty much makes it even weirder. It should work live, by all means...

Would you be able to upload the baseband?

K4KDR, 2021-11-06:
Thank you for the follow-up!

For the past few days, I have been getting perfect images after recording a baseband & then using the Offline Processing tab.  Tonight, however, even though the SNR was fine, the recording did not produce any signal lock or images. 

[image]

... however, this is the first time I have tried an 'i8' recording - all of my previous successful attempts were with the 'i16' format.  

Tonight's i8 recording is available at:

https://www.qsl.net/k/k4kdr//iq/meteor-m2/00-00-27_2500000SPS_137100000Hz.zip

Thanks very much!

-Scott,  K4KDR

K4KDR, 2021-11-06:
[v0.0.38 / master branch / build: 01a8276, built from source]

Running 'Live Processing' with Airspy-R2 @ 2.5M Sample Rate:

**SNR** & **PEAK SNR** values displayed correctly - thank you!!

[image]

Aang23, 2021-11-08:
Awesome! As a note, https://github.com/altillimity/SatDump/commit/a92e595a0993430b5e43625b25700b09849839f8 should also affect this issue, in theory.
I've finally figured that one out after LazzSnazz had issues on GOES-R HRIT.

K4KDR, 2021-11-09:
Very good - will continue to update to the latest.

Many thanks!

andriux26, 2021-11-18:
Who will help with the command need live build
-DBUILD_LIVE=ON
cmake -DCMAKE_BUILD_TYPE=Release ..                             # MacOS
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr .. #
How correct enter

Aang23, 2021-11-19:
> Who will help with the command need live build -DBUILD_LIVE=ON cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr .. # How correct enter linux

I replied in your issue. Please carry on in there if you have any other questions, this is not what this one is about...

Aang23, 2021-11-19:
By the way @K4KDR, have you ever had that issue again? If not, I think I will close this.

andriux26, 2021-11-19:
Fanks for you  hepl i launch satdump live you can closed

K4KDR, 2021-11-19:
No, with the latest builds I seem to be ok on that.  Thanks so much - ok to close!

closed 2021-11-19 by Aang23