SatDump/SatDump #60: WIN32: Illegal use of gmtime()

gvanem, 2021-11-04:
Looking at the CI build log for Windows (e.g. [here](https://pipelines.actions.githubusercontent.com/yDYBoxjbMoNMJbGYB2J6ZLwIoDWHO6sBi0GC5kzZ0We1S3kiu3/_apis/pipelines/1/runs/1769/signedlogcontent/2?urlExpires=2021-11-04T10%3A08%3A17.8017988Z&urlSigningMethod=HMACV1&urlSignature=0cP4rynH%2BagGndslK6SoBizRSr6GbQDZbTMKun%2Bapk8%3D)) there is this important warning from `cl`:
``` 
D:\a\SatDump\SatDump\src-core\libs\predict\julian_date.c(91,20): 
warning C4020: 'gmtime': too many actual parameters [D:\a\SatDump\build\src-core\satdump_core.vcxproj]
``` 

And using `clang-cl`, it's an error. I've patched it like this using the Secure functions:
```diff
--- a/src-core/libs/predict/julian_date.c  2021-10-25 14:22:24
+++ b/src-core/libs/predict/julian_date.c 2021-11-04 11:04:05
@@ -20,7 +20,7 @@
        //get UTC time, interpret resulting tm as a localtime
        struct tm timeinfo_gmt;
 #ifdef _WIN32
-       memcpy(&timeinfo_gmt, gmtime(&curr_time), sizeof(struct tm));
+       gmtime_s(&timeinfo_gmt, &curr_time);
 #else
        gmtime_r(&curr_time, &timeinfo_gmt);
 #endif
@@ -29,7 +29,7 @@
        //get localtime, interpret resulting tm as localtime
        struct tm timeinfo_local;
 #ifdef _WIN32
-       memcpy(&timeinfo_local, localtime(&curr_time), sizeof(struct tm));
+       localtime_s(&timeinfo_local, &curr_time);
 #else
        localtime_r(&curr_time, &timeinfo_local);
 #endif
@@ -88,7 +88,7 @@
        //add number of seconds since julian start day to the julian start day, get current time_t
        struct tm timeinfo;
 #ifdef _WIN32
-       gmtime(&ret_time, &timeinfo);
+       gmtime_s(&timeinfo, &ret_time);
 #else
        gmtime_r(&ret_time, &timeinfo);
 #endif
```

Aang23, 2021-11-17:
Indeed, pretty good idea to fix that.
Shall I apply your patch and push or do you wish to open a PR?

gvanem, 2021-11-17:
> Shall I apply your patch and push ...

Please do. I suck at git.

commit 92bb07ad (2022-01-27): applied gvanem's solution, close #60
closed 2022-01-27 by ZbychuButItWasTaken