SatDump/SatDump #62: QPSK demodulator not detecting signal

benb0jangles, 2021-11-05:
Hi Aang I am trying to use Live Decoding on Satdump for Meteor-M2 however I am unable to get good signal for the demodulator to start working. If I use SDR++ Meteor demodulator on the same machine it works just fine. In satdump, the demoduator only gets weak signal when satellite overhead.

Setup:
Airspy mini
QFH
Jetson Nano with Ubuntu 18.04

I try adjusting gain, enabling or disabling DC Lock, Adjusting the scale sliders, etc but I am unable to find success.

Best wishes

Aang23, 2021-11-05:
That's definitely not supposed to happen.
Do you have a screenshot? Could you perhaps provide some baseband? (Since on my end, it does seem to lock fine)

Thanks

benb0jangles, 2021-11-05:
[image]

benb0jangles, 2021-11-05:
[image]

benb0jangles, 2021-11-05:
https://user-images.githubusercontent.com/15876842/140502837-9b12976f-e9e4-4ec9-bcb8-0b3efb982367.mp4

benb0jangles, 2021-11-05:
[image]

jtop system stats seem ok when running satdump

benb0jangles, 2021-11-07:
Log from this morning Satdump Meteor M2 attempt:
```

~/satdump/build$ ./satdump-ui
[11/07/21 - 08:07:14] (I) Usage : ./satdump-ui [downlink] [input_level] [input_file] [output_level] [output_file_or_directory] [additional options as required]
[11/07/21 - 08:07:14] (I) Extra options : -samplerate [baseband_samplerate] -baseband_format [f32/i16/i8/w8] -dc_block -iq_swap
[11/07/21 - 08:07:14] (I)    _____       __  ____                      
[11/07/21 - 08:07:14] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[11/07/21 - 08:07:14] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[11/07/21 - 08:07:14] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[11/07/21 - 08:07:14] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[11/07/21 - 08:07:14] (I)                                     /_/      
[11/07/21 - 08:07:14] (I) Starting SatDump v0.0.37
[11/07/21 - 08:07:14] (I) 
[11/07/21 - 08:07:14] (I) Loading config from file /usr/share/satdump/config.json
[11/07/21 - 08:07:14] (E) No config file found! Using defaults.
[11/07/21 - 08:07:14] (D) Registered modules (90) : 
[11/07/21 - 08:07:14] (D)  - angels_argos
[11/07/21 - 08:07:14] (D)  - aqua_airs
[11/07/21 - 08:07:14] (D)  - aqua_amsu
[11/07/21 - 08:07:14] (D)  - aqua_ceres
[11/07/21 - 08:07:14] (D)  - aqua_db_decoder
[11/07/21 - 08:07:14] (D)  - aura_omi
[11/07/21 - 08:07:14] (D)  - bpsk_demod
[11/07/21 - 08:07:14] (D)  - ccsds_analyzer
[11/07/21 - 08:07:14] (D)  - elektro_arktika_msugs
[11/07/21 - 08:07:14] (D)  - elektro_arktika_rdas_decoder
[11/07/21 - 08:07:14] (D)  - elektro_arktika_tlm_demod
[11/07/21 - 08:07:14] (D)  - elektro_lrit_data_decoder
[11/07/21 - 08:07:14] (D)  - eos_modis
[11/07/21 - 08:07:14] (D)  - falcon_decoder
[11/07/21 - 08:07:14] (D)  - fengyun_ahrpt_decoder
[11/07/21 - 08:07:14] (D)  - fengyun_erm
[11/07/21 - 08:07:14] (D)  - fengyun_mersi1
[11/07/21 - 08:07:14] (D)  - fengyun_mersi2
[11/07/21 - 08:07:14] (D)  - fengyun_mersill
[11/07/21 - 08:07:14] (D)  - fengyun_mpt_decoder
[11/07/21 - 08:07:14] (D)  - fengyun_mwhs
[11/07/21 - 08:07:14] (D)  - fengyun_mwhs2
[11/07/21 - 08:07:14] (D)  - fengyun_mwri
[11/07/21 - 08:07:14] (D)  - fengyun_mwts
[11/07/21 - 08:07:14] (D)  - fengyun_mwts2
[11/07/21 - 08:07:14] (D)  - fengyun_mwts3
[11/07/21 - 08:07:14] (D)  - fengyun_satid
[11/07/21 - 08:07:14] (D)  - fengyun_svissr_decoder
[11/07/21 - 08:07:14] (D)  - fengyun_svissr_image_decoder
[11/07/21 - 08:07:14] (D)  - fengyun_virr
[11/07/21 - 08:07:14] (D)  - fengyun_waai
[11/07/21 - 08:07:14] (D)  - fengyun_windrad
[11/07/21 - 08:07:14] (D)  - fsk_demod
[11/07/21 - 08:07:14] (D)  - gk2a_lrit_data_decoder
[11/07/21 - 08:07:14] (D)  - goes_gvar_decoder
[11/07/21 - 08:07:14] (D)  - goes_gvar_image_decoder
[11/07/21 - 08:07:14] (D)  - goes_lrit_data_decoder
[11/07/21 - 08:07:14] (D)  - jason3_amr2
[11/07/21 - 08:07:14] (D)  - jason3_decoder
[11/07/21 - 08:07:14] (D)  - jason3_lpt
[11/07/21 - 08:07:14] (D)  - jason3_poseidon
[11/07/21 - 08:07:14] (D)  - jpss_atms
[11/07/21 - 08:07:14] (D)  - jpss_cris
[11/07/21 - 08:07:14] (D)  - jpss_satid
[11/07/21 - 08:07:14] (D)  - jpss_viirs
[11/07/21 - 08:07:14] (D)  - meteor_hrpt_decoder
[11/07/21 - 08:07:14] (D)  - meteor_hrpt_demod
[11/07/21 - 08:07:14] (D)  - meteor_lrpt_decoder
[11/07/21 - 08:07:14] (D)  - meteor_msumr
[11/07/21 - 08:07:14] (D)  - meteor_msumr_lrpt
[11/07/21 - 08:07:14] (D)  - meteor_mtvza
[11/07/21 - 08:07:14] (D)  - metop_admin_msg
[11/07/21 - 08:07:14] (D)  - metop_ahrpt_decoder
[... 341 lines cut ...]
[11/07/21 - 08:39:50] (I) Progress inf%, SNR : 0.000000dB, Peak SNR: 0.000000dB
[11/07/21 - 08:39:50] (I) Progress inf%, Viterbi BER : 26.757812%, Lock : NOSYNC
[11/07/21 - 08:40:00] (I) Progress inf%, SNR : 0.000000dB, Peak SNR: 0.000000dB
[11/07/21 - 08:40:00] (I) Progress inf%, Viterbi BER : 25.195312%, Lock : NOSYNC
[11/07/21 - 08:40:10] (I) Progress inf%, SNR : 0.000000dB, Peak SNR: 0.000000dB
[11/07/21 - 08:40:10] (I) Progress inf%, Viterbi BER : 26.562500%, Lock : NOSYNC
[11/07/21 - 08:40:20] (I) Progress inf%, SNR : 0.000000dB, Peak SNR: 0.000000dB
[11/07/21 - 08:40:20] (I) Progress inf%, Viterbi BER : 26.757812%, Lock : NOSYNC
[11/07/21 - 08:40:30] (I) Progress inf%, Viterbi BER : 25.000000%, Lock : NOSYNC
[11/07/21 - 08:40:30] (I) Progress inf%, SNR : 0.000000dB, Peak SNR: 0.000000dB
[11/07/21 - 08:40:40] (I) Progress inf%, SNR : 0.000000dB, Peak SNR: 0.000000dB
[11/07/21 - 08:40:40] (I) Progress inf%, Viterbi BER : 26.171875%, Lock : NOSYNC
[11/07/21 - 08:40:41] (I) Stop processing
[11/07/21 - 08:40:41] (I) Stop processing
[11/07/21 - 08:40:41] (I) Demodulation finished
[11/07/21 - 08:40:41] (I) Stopping SDR
[11/07/21 - 08:40:41] (I) Stop FFT
[11/07/21 - 08:40:41] (I) lock
[11/07/21 - 08:40:41] (I) FFT exit
[11/07/21 - 08:40:41] (I) Saving settings...
[11/07/21 - 08:40:41] (I) Saving settings to file /home/benja2/.config/satdump/settings.json
[11/07/21 - 08:40:41] (I) Destroying objects
[11/07/21 - 08:40:41] (I) Live exited!
[11/07/21 - 08:40:45] (I) Exiting!
```

Aang23, 2021-11-08:
Sorry for the late answer. I've had similar reports that lead to this fix https://github.com/altillimity/SatDump/commit/a92e595a0993430b5e43625b25700b09849839f8, which I think also affected LRPT as it pretty much always go through the resampler...

funnymdzz, 2021-11-11:
Hello when I receive the Elektro L3 it seemed to be the same problem. The RS is red and the BER is jumped from green and red

https://user-images.githubusercontent.com/25901780/141256512-2dd271e8-42b1-4546-81e0-df8bdd98b347.mp4

Aang23, 2021-11-11:
That is definitely not the same issue, the BPSK demod locked just fine.
Could you perhaps try recording baseband and seeing if that works? 

Also, are you running the latest APK from Github CI?

benb0jangles, 2021-11-11:
@Aang23 I am noticing also during my build stage:
`cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_LIVE=ON -DCMAKE_INSTALL_PREFIX=/usr ..`
Volk message:
`Performing Test VOLK_HAS_volk_32fc_x2_add_32fc - Failed`
But still compiles

So perhaps I will try to build Volk from source and see what happens.

Aang23, 2021-11-11:
About this error when building, that is expected if you are running Volk <=1.3. I do this test to provide a fallback implementation if Volk does not provide it.
Failing or passing doesn't really matter at all, some (unused) code would just be a bit slower.

No, volk_config should be left where it is.

That's weird when compiling Volk though... May be an ARM-specific issue, but I'd advise opening up an issue in Volk. Not a SatDump issue by any means.

benb0jangles, 2021-11-11:
Ah ok :)

I compiled https://github.com/altillimity/SatDump/issues/62#issuecomment-963602877 fix - Just waiting for Meteor Pass in hour or so.

Aang23, 2021-11-11:
Good luck with that pass! Actually I may as well do it too, just gotta take the QFH out.

benb0jangles, 2021-11-11:
No success. Demodulator was still the same as above.

benb0jangles, 2021-11-12:
Testing this morning again but using Macbook Pro:

Still no demodulation of Meteor-M2

https://user-images.githubusercontent.com/15876842/141438614-e60b56eb-cf14-4e42-abca-a5613b43e5af.mp4

Differences in the Log between Jetson Nano & Macbook are Macbook has Volk Spiral, and SNR is better

Log:
```
[11/12/21 - 08:36:23] (D) Starting livedemod...
[11/12/21 - 08:36:23] (I) meteor_m2_lrpt
[11/12/21 - 08:36:23] (I) Starting live pipeline METEOR M2 LRPT
[11/12/21 - 08:36:23] (I) Module qpsk_demod
[11/12/21 - 08:36:23] (D) Parameters :
[11/12/21 - 08:36:23] (D)    - agc_rate : 0.1
[11/12/21 - 08:36:23] (D)    - baseband_format : f32
[11/12/21 - 08:36:23] (D)    - buffer_size : 8192
[11/12/21 - 08:36:23] (D)    - costas_bw : 0.004
[11/12/21 - 08:36:23] (D)    - dc_block : 1
[11/12/21 - 08:36:23] (D)    - rrc_alpha : 0.6
[11/12/21 - 08:36:23] (D)    - rrc_taps : 361
[11/12/21 - 08:36:23] (D)    - samplerate : 1000000
[11/12/21 - 08:36:23] (D)    - symbolrate : 72000
[11/12/21 - 08:36:23] (I) Module meteor_lrpt_decoder
[11/12/21 - 08:36:23] (D) Parameters :
[11/12/21 - 08:36:23] (D)    - baseband_format : f32
[11/12/21 - 08:36:23] (D)    - dc_block : 1
[11/12/21 - 08:36:23] (D)    - diff_decode : 0
[11/12/21 - 08:36:23] (D)    - samplerate : 1000000
[11/12/21 - 08:36:23] (T) Volk has the spiral kernel, using it!
[11/12/21 - 08:36:23] (D) Starting SDR...
[11/12/21 - 08:36:23] (D) Device parameters airspy:
[11/12/21 - 08:36:23] (D)    - bias : 1
[11/12/21 - 08:36:23] (D)    - gain : 22
[11/12/21 - 08:36:23] (I) Opened Airspy device!
[11/12/21 - 08:36:23] (I) Samplerate 1000000.000000
[11/12/21 - 08:36:23] (I) Frequency 137100000.000000
[11/12/21 - 08:36:23] (D) Input SPS : 13.888889
[11/12/21 - 08:36:23] (D) Resample : 1
[11/12/21 - 08:36:23] (D) Samplerate : 210000.000000
[11/12/21 - 08:36:23] (D) Dec factor : 4.761905
[11/12/21 - 08:36:23] (D) Final SPS : 2.916667
[11/12/21 - 08:36:23] (I) Start processing...
[11/12/21 - 08:36:23] (I) Using input baseband 
[11/12/21 - 08:36:23] (I) Demodulating to /Users/benbo/Desktop/Meteor/meteor_m2_lrpt.soft
[11/12/21 - 08:36:23] (I) Buffer size : 40960
[11/12/21 - 08:36:23] (I) Start processing...
[11/12/21 - 08:36:23] (I) Using input symbols 
[11/12/21 - 08:36:23] (I) Decoding to /Users/benbo/Desktop/Meteor/meteor_m2_lrpt.cadu
[11/12/21 - 08:36:30] (I) Progress nan%, SNR : 0.000000dB, Peak SNR: 0.000000dB
[11/12/21 - 08:36:30] (I) Progress nan%, Viterbi BER : 26.757812%, Lock : NOSYNC
[11/12/21 - 08:36:40] (I) Progress nan%, SNR : 0.000000dB, Peak SNR: 1.052344dB
[11/12/21 - 08:36:40] (I) Progress nan%, Viterbi BER : 26.757812%, Lock : NOSYNC
[11/12/21 - 08:36:50] (I) Progress nan%, SNR : 0.000000dB, Peak SNR: 1.052344dB
[11/12/21 - 08:36:50] (I) Progress nan%, Viterbi BER : 24.218750%, Lock : NOSYNC
[11/12/21 - 08:37:00] (I) Progress nan%, SNR : 0.000000dB, Peak SNR: 1.052344dB
[11/12/21 - 08:37:00] (I) Progress nan%, Viterbi BER : 26.562500%, Lock : NOSYNC
[11/12/21 - 08:37:10] (I) Progress nan%, SNR : 0.000000dB, Peak SNR: 1.052344dB
[11/12/21 - 08:37:10] (I) Progress nan%, Viterbi BER : 26.171875%, Lock : NOSYNC
[11/12/21 - 08:37:20] (I) Progress nan%, SNR : 2.274144dB, Peak SNR: 2.628546dB
[11/12/21 - 08:37:20] (I) Progress nan%, Viterbi BER : 26.562500%, Lock : NOSYNC
[11/12/21 - 08:37:30] (I) Progress nan%, Viterbi BER : 26.757812%, Lock : NOSYNC
[11/12/21 - 08:37:30] (I) Progress nan%, SNR : 0.000000dB, Peak SNR: 2.628546dB
[11/12/21 - 08:37:40] (I) Progress nan%, SNR : 0.000000dB, Peak SNR: 2.628546dB
[11/12/21 - 08:37:40] (I) Progress nan%, Viterbi BER : 25.195312%, Lock : NOSYNC
[11/12/21 - 08:37:50] (I) Progress nan%, SNR : 0.000000dB, Peak SNR: 2.628546dB
[11/12/21 - 08:37:50] (I) Progress nan%, Viterbi BER : 25.585938%, Lock : NOSYNC
[11/12/21 - 08:38:00] (I) Progress nan%, SNR : 0.000000dB, Peak SNR: 2.628546dB
[11/12/21 - 08:38:00] (I) Progress nan%, Viterbi BER : 26.171875%, Lock : NOSYNC
[... 20 lines cut ...]
[11/12/21 - 08:39:50] (I) Progress nan%, SNR : 0.000000dB, Peak SNR: 2.628546dB
[11/12/21 - 08:39:50] (I) Progress nan%, Viterbi BER : 26.757812%, Lock : NOSYNC
[11/12/21 - 08:40:00] (I) Progress nan%, SNR : 0.000000dB, Peak SNR: 2.628546dB
[11/12/21 - 08:40:00] (I) Progress nan%, Viterbi BER : 26.171875%, Lock : NOSYNC
[11/12/21 - 08:40:10] (I) Progress nan%, SNR : 0.000000dB, Peak SNR: 2.628546dB
[11/12/21 - 08:40:10] (I) Progress nan%, Viterbi BER : 26.562500%, Lock : NOSYNC
[11/12/21 - 08:40:20] (I) Progress nan%, SNR : 0.000000dB, Peak SNR: 2.628546dB
[11/12/21 - 08:40:20] (I) Progress nan%, Viterbi BER : 25.585938%, Lock : NOSYNC
[11/12/21 - 08:40:30] (I) Progress nan%, SNR : 0.000000dB, Peak SNR: 2.628546dB
[11/12/21 - 08:40:30] (I) Progress nan%, Viterbi BER : 26.171875%, Lock : NOSYNC
[11/12/21 - 08:40:40] (I) Progress nan%, SNR : 0.000000dB, Peak SNR: 2.628546dB
[11/12/21 - 08:40:40] (I) Progress nan%, Viterbi BER : 26.367188%, Lock : NOSYNC
[11/12/21 - 08:40:44] (I) Stop processing
[11/12/21 - 08:40:44] (I) Stop processing
[11/12/21 - 08:40:44] (I) Demodulation finished
[11/12/21 - 08:40:44] (I) Stopping SDR
[11/12/21 - 08:40:44] (I) Stop FFT
[11/12/21 - 08:40:44] (I) lock
[11/12/21 - 08:40:44] (I) FFT exit
[11/12/21 - 08:40:44] (I) Saving settings...
[11/12/21 - 08:40:44] (I) Saving settings to file /Users/benbo/.config/satdump/settings.json
[11/12/21 - 08:40:44] (I) Destroying objects
[11/12/21 - 08:40:44] (I) Live exited!

```

benb0jangles, 2021-11-16:
@Aang23 as an update:
Today I recorded the .s file for a Meteor-M2 pass using SDR++ and SDR++ Meteor Demodulator. 
Offline Processing this file using satdump works and I have satellite image as expected.

So perhaps the satdump live processing demodulator is not successful just yet?

Aang23, 2021-11-28:
@benb0jangles sorry for the late answer.

It's possible something is going wrong in the demodulator, but others and I have been unable to reproduce the issue. Are you able to provide a baseband sample? Really the only way I may be able to figure it out with this kind of issue.

benb0jangles, 2021-11-29:
@Aang23 I have recorded baseband using satdump + these settings:
Device: Airspy
Frequency: 137.1Mhz
Samplerate: 1000000Hz
Format: i16

The .i16 file is over 3GB in size so I must wait a couple of hours before I can complete the upload to gdrive

Here are screen recordings during baseband recording

https://user-images.githubusercontent.com/15876842/143832131-a9c9c04a-7422-483a-aa1f-678da345c081.mp4

https://user-images.githubusercontent.com/15876842/143832194-a936a41a-5b36-4ef9-9e51-a3cdcd89f923.mp4

[image]

:)

benb0jangles, 2021-11-29:
@Aang23 

Here is the baseband recording:
[](https://drive.google.com/file/d/1Qpd_0O7nU2MEg9dEbdN0wcYvpyJdkDGp/view?usp=sharing)

Aang23, 2021-11-29:
@benb0jangles well, that settles it.
The Airspy cannot do 1MSPS. The Mini can do **only** 3MSPS and 6MSPS, and the R2 2.5MSPS and 10MSPS. Every other samplerate will not work and cause this kind of issues!

benb0jangles, 2021-11-29:
So I guess I will use RTL SDR for LRPT, then Airspy for HRPT

K4KDR, 2021-11-29:
LRPT works very well on the Airspy-R2 at the 2.5M sample rate & should do well on the Mini at 3M.     If the computer can handle the higher sample rates for those devices, 10M or 6M will work great, too!

benb0jangles, 2021-12-02:
@Aang23 so this morning i tried LRPT using 3M with aIrspy mini and result is improvement. The demodulator is now functioning and SYNC is acheived.
[image]

Signal is not as strong as using SDR# or GQRX or similar software, but I think still capable. For this morning pass I had to contend with NOAA15 and POCSAG interference. I use a 137mhz Bandpass filter, but I think I will remove it tomorrow because the filter lowers the db by 30%. I am hoping stronger signal will create better image, this mornings image was terrible but it was the first one using Satdump live so also success :)

I like that it is working on 5v Jetson Nano 2GB it is amazing now to think that full live LRPT - Signal - Demodulation - Decoding - Processing is functional on cheap ARM device.

My future goals will of course to push and see if it can also do Live HRPT at 6M once I finish making my DIY parabolic antenna, and the rain stops.

Good luck with development, I think you have powerful software in Satdump :)

benb0jangles, 2021-12-04:
Samples:
[image]

[image]

[image]

benb0jangles, 2021-12-06:
Recorded using SDR++ on JEtson Nano with Meteor Plugin, then SAtdump offline Processing of 100MB .s file
[image]

Aang23, 2021-12-18:
Nice results! I guess I can close this now!

closed 2021-12-18 by Aang23