SatDump/SatDump #65: Feature Request: CI examples.

errolt, 2021-11-10:
Hi Aang32,

Please add more examples for command line usage. 
For example, it is possible to do Live processing from command line? What is the command?

Please consider adding a feature to the GUI that prints to the console, when you click "start", the CI command that one can use to run the same pipeline from the command line.

errolt, 2021-11-10:
Also, an example of a config file for the satdump-ingestor.
This is hat I could figure out from the source code, but I don't know what the http_server entry is for:
`{
    "sdr_type": "",
    "sdr_settings": {
        "gain": "34",
        "bias": "1"
    },
    "samplerate":"1000000",
    "frequency":"1691000",
    "pipeline":"elektro_lrit",
    "output_folder":"./image",
    "http_server":""
}
`

Aang23, 2021-11-11:
Hi @errolt !

I agree on everything. I'd even extend outside of CLI documentation.

SatDump's documentation is pretty much nonexistent and that's honestly an issue. There are so many features and things not everyone may know getting into it... I very often get some questions which should be documented.

Documentation's coming, and it is something I've got on my to-do for a while... Though about the ingestor, you may find some on Github's Wiki tab (should answer your questions).

About live-processing from the CLI, right now it's mostly done with the ingestor, but that is of course not the only usecase. Better support for launching it directly from a single command is planned. That is, alongside allowing scheduling of live-processing and recording some of the things I'd like to work on soon enough.

errolt, 2021-11-11:
Hi Aang23
That is fine. I know that code comes first, documentation later.
I have figured out most of what I need by looking at the source.

Aang23, 2022-09-16:
@errolt is there enough now? Or do you wish there was more?

Aang23, 2022-09-21:
I will close this - should be covered and more is upcoming shortly.

closed 2022-09-21 by Aang23