SatDump/SatDump #66: Android samplerate problem

funnymdzz, 2021-11-12:
I want to use 6000000HZ samplerate for fy-2h decode on android  but when I set the samplerate to 6000000HZ the output was like this even though set it to 10000000HZ. the BPSK demod constellation is slow.use a lower samplerate like 2500000HZ is working but the SNR is reduced.
[image]

funnymdzz, 2021-11-12:
SDR is Airspy One

funnymdzz, 2021-11-12:
almost forget it is live processing

Aang23, 2021-11-12:
What device are you using?

Aang23, 2021-11-12:
This honestly just looks like sampledrop due to an older drive or too much stuff running in the background to me. Is that an airspy R2 or Mini though?

funnymdzz, 2021-11-12:
> This honestly just looks like sampledrop due to an older drive or too much stuff running in the background to me. Is that an airspy R2 or Mini though?

it is a airspy one and in the pc it is no problem when I use high samplerate

Aang23, 2021-11-17:
Yes, but it's not because your PC does it fine your phone will. They are different hardware and for many reasons it could not be performing the same.

closed 2022-01-27 by ZbychuButItWasTaken