SatDump/SatDump #67: Projection: Custom labels don't support UTF-8

jpjonte, 2021-11-13:
Hi,

I just tried adding "Lübeck" as a custom map label and discovered that the ü isn't rendered properly.
As far as I can tell, this is a limitation of CImg.

Aang23, 2021-11-17:
Hi!
Indeed, as of now everything is done as ASCII.. I guess UTF-8 support would probably be a good thing to consider, but may perhaps require some more work around CImg to get it working.

Aang23, 2022-01-05:
Update : 
Since the image / text handling is now custom, this will be easier to implement.

ZbychuButItWasTaken, 2022-09-16:
currently labels aren't implemented anymore but we will work on them. they will include utf-8 support

ZbychuButItWasTaken, 2023-02-24:
This issue was not forgotten btw, it just turned out to be something more complicated, as we will have to implement a font engine to the image lib. However we will probably use imgui's impl, so it shouldn't take long!

AlexandreRouma, 2023-02-24:
ImGui supports UTF-8 natively but the issue is the font atlas gets quite big when adding CJK stuff and it causes issues on older devices with a lower texture size limitation (especially common on Android, even for mid-range devices).

ZbychuButItWasTaken, 2023-02-24:
yes I know it does, but our image lib (which is what the whole issue is about) doesn't.. (For now we used a pre-renderd font image which obviously is not optimal. We are gonna switch to rendering the font from a ttf)

We use imgui with some custom glyph ranges to help with performance and it seems to work quite well (yes we run into issues on android lol)

Aang23, 2023-02-24:
Even on some slightly older laptops this was killing the font... Either way, in this case it's for rendering to images internally so a while different story.

Another reason it took so long to get to this is simply a certain rewrite.

jpjonte, 2023-02-24:
Appreciate the insight. Font rendering can be a bitch.