SatDump/SatDump #68: live elektro L TLM fail

miguelvrr, 2021-11-14:
Hi, I have already the newest build for win with the live processing aparently working. Satdump is able to open my airspy but the elektro L TLM signal is non present. I can't decode anything and the signal stays on 0.0 Any idea how to configure the software to receive this?? All help is welcome! Thx

Aang23, 2021-11-16:
Hi, do you have a screenshot of the issue?

miguelvrr, 2021-11-20:
Hi sir,
Yes i sent you the prints in attachment. Still not able to decode any signal.
Is the sat still active and transmitting in 1693? The 1544 beacon seems to be working fine.

0[image]
[image]

errolt, 2021-11-23:
@miguelvrr 
For me it works. But my signal is weak, so locking is a problem.
[image]

Aang23, 2021-11-28:
https://github.com/altillimity/SatDump/commit/6d0de5d8255e841b8f7cc230efce39927748d598 should have helped a lot with this issue.

closed 2022-01-27 by ZbychuButItWasTaken