SatDump/SatDump #71: 32-bit support?

gvanem, 2021-11-19:
I'd like to see 32-bit (`x86`) support in SatDump programs.  Since most of my external packages are for 32-bit. 
(I guess that's the situation for many developers too).

I've built SatDump for `x86` on Windows with only this patch:
```diff
--- a/src-core/libs/aec/gcc_builtins.c 2021-10-25 14:22:20
+++ b/src-core/libs/aec/gcc_builtins.c 2021-11-19 19:53:19
@@ -1,8 +1,9 @@
 /*
     MSVC lacks some builtin functions GCC has
-    From https://gist.github.com/pps83/3210a2f980fd02bb2ba2e5a1fc4a2ef0
+    Edited from https://gist.github.com/pps83/3210a2f980fd02bb2ba2e5a1fc4a2ef0
+    Added support for MSVC-x86 ('_WIN64' is not defined).
 */
-#ifdef _MSC_VER
+#if defined(_MSC_VER) && !defined(__clang__)
 #include <intrin.h>
 #include <stdint.h>

@@ -14,7 +15,15 @@

 int __builtin_ctzll(unsigned long long x) {
     unsigned long ret;
+
+#ifdef _WIN64
     _BitScanForward64(&ret, x);
+#else
+    if (_BitScanForward(&ret, (uint32_t)x) == 0) { // low 32 bits are all zero.
+      _BitScanForward(&ret, (uint32_t)(x >> 32));
+      ret += 32;
+    }
+#endif
     return (int)ret;
 }

@@ -23,17 +32,21 @@
 }

 int __builtin_clz(unsigned x) {
-    //unsigned long ret;
-    //_BitScanReverse(&ret, x);
-    //return (int)(31 ^ ret);
     return (int)__lzcnt(x);
 }

 int __builtin_clzll(unsigned long long x) {
-    //unsigned long ret;
-    //_BitScanReverse64(&ret, x);
-    //return (int)(63 ^ ret);
+#ifdef _WIN64
     return (int)__lzcnt64(x);
+#else
+    unsigned long ret;
+    // Scan the high 32 bits.
+    if (_BitScanReverse(&ret, (uint32_t)(x >> 32)))
+       return (63 ^ (ret + 32));
+    // Scan the low 32 bits.
+    _BitScanReverse(&ret, (uint32_t)x);
+    return (63 ^ (int)ret);
+#endif
 }

 int __builtin_clzl(unsigned long x) {
@@ -49,4 +62,5 @@
     return __builtin_clzll(x);
 }
 #endif
-#endif
\ No newline at end of file
+#endif // _MSC_VER) && !__clang__
```

Added: 
* clean-out old code.
* `clang-cl.exe` (`__clang__`) defines `_MSC_VER` too. But it already has these `__builtin_x()` functons.

But adapting `CMakeLists.txt` , I've no idea how to adapt for `x86` (I use a home-brew GNU `Makefile`).

So how about it?

Aang23, 2021-11-28:
I'm definitely OK with that.
Can you confirm this does not modify any behavior of the code utilizing those functions? If not, I'll probably give it a quick check a bit later and push that.

gvanem, 2021-11-29:
> Can you confirm this does not modify any behavior of the code utilizing those functions? 

I'm 99% confident it does not. But a CI-build could verify that.
Besides, it's very similar to what's in [src-core/spdlog/fmt/bundled/format.h](https://github.com/altillimity/SatDump/blob/master/src-core/spdlog/fmt/bundled/format.h).

errolt, 2021-12-05:
As a side note, it looks like geos_gvar doesn't run in 32 bit mode, at least on my Raspberry Pi 4GB with the official OS, which only ships in 32 bit.
Running 64 bit Ubuntu works, it does use VM when creating false color image, but doesn't crash.

32bit gives me this, without touching swap:
> [CImg] *** CImgInstanceException *** [instance(0,0,0,0,(nil),non-shared)] CImg<unsigned short>::CImg(): Failed to allocate memory (753.3 Mio) for image (20836,18956,1,1).
terminate called after throwing an instance of 'cimg_library::CImgInstanceException'
  what():  [instance(0,0,0,0,(nil),non-shared)] CImg<unsigned short>::CImg(): Failed to allocate memory (753.3 Mio) for image (20836,18956,1,1).
Aborted

gvanem, 2021-12-05:
>  Failed to allocate memory (753.3 Mio)

753 MByte!!?? This is unrelated to this issue. But too little free RAM.

> Don't know if this will be an issue on 32bit x86 too..

I've never seen the `CImgInstanceException` on x86. Only `realloc()` failing in the Waterfall. But again, this is unrelated to 
this issue.

errolt, 2021-12-05:
I watched the ram, had 2GB free. As I said, same thing does work on an 64bit OS on the same hardware. From what I read there is, on x86(don't know about arm), a 4GB hard limit on Virtual Memory. This might be hit before 4GB real memory is used? 

@gvanem have you tried receiving goes_gvar on x86 yet? Don't know about the other GOES sats, but for GOES13 the main image is a gray scale 400 megapixel image, 200MB on disk after png compression, and SatDump uses that 400MP image and a 25MP image to build another 400MP false color image, while receiving the next set of images containing four 25MP images and another 400MP image. In other words, SatDump can have three 400MP images and 5 25MP images in ram, all at the same time... Might be a hard push on a 4GB limit imposed by 32bit?

closed 2022-09-16 by ZbychuButItWasTaken