SatDump/SatDump #77: enancement: Best out of 3 voting for block_id in goes_gvar

errolt, 2021-12-09:
For goes_gvar decoding, the block_header is repeated 3 times for redundancy.

Ideally one would CRC the header, and if it fails, try the next header. But as we only use the block_id I suggest a "best out of three vote" for block_id. This would reduce the chance of a wrong block_id due to a bit flip, and image data ending up in the wrong image. 
```
// Parse main header
  PrimaryBlockHeader block_header1 = *((PrimaryBlockHeader *)&frame[8]);
  PrimaryBlockHeader block_header2 = *((PrimaryBlockHeader *)&frame[8+30]);
  PrimaryBlockHeader block_header3 = *((PrimaryBlockHeader *)&frame[8+60]);
  int block_id=0;
  if(block_header1.block_id==block_header2.block_id)
  {
          block_id=block_header1.block_id;
  }
  else if(block_header2.block_id==block_header3.block_id)
  {
          block_id=block_header2.block_id;
  }
  else if(block_header3.block_id==block_header1.block_id)
  {
          block_id=block_header1.block_id;
  }
```

Aang23, 2021-12-10:
Indeed, that would be a lot better and something I should have done!

My reason for *not* parsing CRCs is that unlike a "real" station, amateur setups will usually have much lower link margins so parsing them would result in discarding most if not all of the data despite only being a few bad pixels.

Do you want to open a PR?

errolt, 2021-12-10:
I do agree, doing CRC would reject a header if there is a fault anywhere in the header, 99% of which we don't care about. 

I can do a PR, but I don't know if that code is optimal. Someone with a statistical background might have a better way? My code just brute-forces it...

Aang23, 2021-12-10:
I would probably just swap it out for the most_common method I have in common/utils.h. I guess another (better, but perhaps overkill) way would be comparing on the bit-level to establish a best match against known block IDs.

Did you actually check how this performed against doing it with a single block header?

Aang23, 2021-12-10:
Another way that I just thought about : It would be possible to check all 3 CRCs, if one passes => Use that header's ID, otherwise if none passes, fallback to the most common ID between all 3.

errolt, 2021-12-10:
That would be best, and if I understand correctly goes crc uses the same polynomial as elektro crc?

I couldn't get it to sum correctly though. I have not worked with CRCs in years, but I remember vaguely that CRCing the whole packet, including the CRC bytes, should give you 0x0000 or 0xffff, which didn't happen here. Not sure why. Maybe gvar crc is not the same as elektro?

errolt, 2021-12-19:
I have implemented CRC, but the "most_common" function does not like it when there are only two results to choose from?
I pass it a vector with the values 203, 11, 11. And debugging I see that, in most_common, the counts[] array contains this:
```
[11]:2
[203]:1
```
But the last line of most_common still returns 203?

My code:
```
                // Parse main header
                std::vector<uint16_t> block_ids;
                PrimaryBlockHeader block_header = *((PrimaryBlockHeader *)&frame[8]);
                block_header.header_crc=~block_header.header_crc;
                if(computeCRC((uint8_t*)&block_header,30)!=0)
                {//CRC failed for first header, try second header
                    block_ids.push_back(block_header.block_id);
                    PrimaryBlockHeader block_header = *((PrimaryBlockHeader *)&frame[8+30]);
                    block_header.header_crc=~block_header.header_crc;
                    if(computeCRC((uint8_t*)&block_header,30)!=0)
                    {//CRC failed for second header, try third header
                        block_ids.push_back(block_header.block_id);
                        PrimaryBlockHeader block_header = *((PrimaryBlockHeader *)&frame[8+60]);
                        block_header.header_crc=~block_header.header_crc;
                        if(computeCRC((uint8_t*)&block_header,30)!=0)
                        {//All headers failed CRC. Use best of 3.
                            block_ids.push_back(block_header.block_id);
                            block_header.block_id=most_common(block_ids.begin(),block_ids.end());
                        }
                    }
                }

```

errolt, 2021-12-19:
Created a pull request.
https://github.com/altillimity/SatDump/pull/79

ZbychuButItWasTaken, 2022-09-17:
closing as it is a PR now!

closed 2022-09-17 by ZbychuButItWasTaken