SatDump/SatDump #80: v0.0.39 not equal MASTER?

K4KDR, 2021-12-19:
Thanks for another round of updates!

I should not 'assume', but when **v0.0.39** was announced I thought that I would be updated to the latest version with a 'git pull' and rebuild from the MASTER branch.

I see that is **not** the case - after 'git pull' & rebuild, MASTER still shows v0.0.38 when run.

Is there any way to know when the MASTER branch repository is the same as the latest announced "release"?

Many thanks!

-Scott,  K4KDR

K4KDR, 2021-12-19:
I believe I found the issue.

Even when downloading & building the v0.0.39 release tar, the app still displays v0.0.38 when run.

So, it seems that it might just be a cosmetic issue with the version stamp within the application.

Thanks!

-Scott

Aang23, 2021-12-22:
Indeed... I somehow entirely forgot to bump the version in the code itself... Oh well, I'll fix that up in Git and it probably isn't such a big deal I'd say. Happens.
Thanks though!

closed 2022-01-27 by ZbychuButItWasTaken