SatDump/SatDump #82: Recording / Live processing from the command line

SV1CAL, 2021-12-22:
Hi Alan, 
This isn't an issue of course :-) ...Just trying to figure out the following :
How can I start SatDump live processing and recording from the command line?  I have been successful with processing recorded files from the cmd line, now wanted to somehow automate live decoding/recording. The idea behind this is to automate (schedule) somehow RX in my own experimental software so that e.g L band passes are live-decoded and X band (except maybe Aqua) are first recorded and processed off line. Thanks and Regards, Michael SV1CAL

Aang23, 2021-12-23:
Hi Michael!

Well, actually I think it is an issue. There is actually no "proper" way to start a live processing pipeline for this kind of usecase, which is definitely something lacking.

Currently, there is the ingestor which can be started from the command-line given a provided JSON configuration, but that is more designed for 24/7 setups on GEOs and similar. Not really for a "pass" system at all.

All I can say is : I should add that to the CLI soon!

SV1CAL, 2021-12-23:
Hi Alan,
Thanks for the feedback ! For the recording and live processing command we could have a recording/processing time  argument . E.g in live decoding the remaining time until LOS could be passed etc.

Anyway, take your time :-)
My best wishes for Christmas and Happy New Year!

Regards,
Michael

mapaknot, 2021-12-24:
Hi Alan.
Thanks, we will wait.
Happy New Year.

errolt, 2022-01-02:
@SV1CAL Hi Michael,
I use these scripts, on a pi, that uses "predict" to find all the passes, schedules the recordings, and uses "timeout" to run an executable for X seconds before killing it. These scripts use wxtoimg to record APT images, but the same framework can be adapted for any satellite.
https://www.instructables.com/Raspberry-Pi-NOAA-Weather-Satellite-Receiver/

SV1CAL, 2022-01-03:
Hi @errolt ,

Thanks ! Very nice work . Will see if I can take some ideas and adapt it somehow.
FYI, I have written a C# application for satellite pass prediction and antenna tracking (L/X band) which automatically launches HDSDR to record a pass programming LimeSDR's IF frequency, sampling rate , gain settings etc.
After a pass, the recorded files are postprocessed offline.
Recording could also be done with SatDump alone (no HDSDR needed)  which would allow for even higher rates (over 60 MSPS). But as far as I now current command line calls don't support recording or live decoding for a given amount of time.

Thanks,
Michael

errolt, 2022-01-03:
@SV1CAL Not my code, just found a long time ago.
There is satdump_recorder in build\tools folder for recording baseband. It would be nice if that supported a timeout.

I do the same on my setup. 
For ATP I record the wav file with rtl_fm, after the pass I process the wav with wxtoimg.
For ElektroL3, which only transmits every 3 hours I recorded baseband with rtl_sdr for 20 minutes, after which I process the baseband with satdump.
For GOES13, which transmits constantly, I use satdump to halfway decode directly as it receives(to save disk space), for 30 minutes, then I kill satdump, rename the file created by satdump, and I restart satdump. On another machine I take that half decoded file and I use satdump to do the rest of the decoding.

As satdump doesn't have a timeout, I just kill the process when I need to. So something like this:
Start satdump
Sleep 20 minutes
Killall satdump

mapaknot, 2022-01-03:
Hi @errolt ,
You have a narrow reception bandwidth.
We need to record at least 3 or 6 broadband channels.
I myself use it on Linux as you predict for NOAA Meteor.
In any case, you need to work in the command line in Win and Linux.

errolt, 2022-01-03:
Hi @mapaknot,
GOES13 is 6Msps, not quite "narrow band".

Michael didn't know there is a baseband recorder tool as part of satdump, and I just pointed him to **build\tools\satdump_recorder**

My advice was on a workaround for not having a timeout in satdump. Process management is unique to every operating system. For powershell on windows there is "Stop-Process -Name ApplicationName" and for plain command line there is "taskkill".

I don't really understand how the bandwidth matters. Except that recording 60Msps would yield 120MB/s to 240MB/s of data that you would need to store somewhere, fast enough. **satdump_recoder** does support compression, but I have not used it before.

Aang23, 2022-01-05:
> Hi Alan, Thanks for the feedback ! For the recording and live processing command we could have a recording/processing time argument . E.g in live decoding the remaining time until LOS could be passed etc.
> 
> Anyway, take your time :-) My best wishes for Christmas and Happy New Year!
> 
> Regards, Michael

You're welcome!
For recording, you can already use the command-line recorder, alongside a command similar to "timeout" on Linux which will send a killsig after a set delay.

Optimally though, would it be for recording or live-processing, you'd never have to kill SatDump that way and rather command it externally somehow, or utilize the raw satdump_core lib.
Still need to settle how I will do things, but as I have started working on tracking features in SatDump, that will be coming soon enough.

Currently, my own L-Band station runs on a custom automated tracking / recording program that deals with the rotator and starting the DSP. It's what I plan on integrating in a tool similar the ingestor and in the UI later on. (My current tool while fully functional on my station is extremely WIP, and not something I'd be comfortable pushing on github!).
I'd guess what you're doing is pretty similar, and could be fixed by something similar to #38 (DDE Control over TCP or something).

> Hi @errolt , You have a narrow reception bandwidth. We need to record at least 3 or 6 broadband channels. I myself use it on Linux, as you predict for NOAA Meteor. In any case, we need to work in the command line on Win and Linux.

What does samplerate have to do with that? Everything should behave the same would it be at 1MSPS or 70.

Aang23, 2022-01-05:
But anyway, my opinion on this topic (and #38) is :
1) A better CLI system to start live-processing and recording is required, both as an user interface and way for users to integrate in existing automation systems.
2) Remote control by common software (UltraTrack, Orbitron, Gpredict, etc) is required as that's a very common way people have been doing this sort of tasks in the past, especially coming from VHF bands and LRPT automation on Windows.
3) An integrated, perhaps more advanced (being integrated directly into SatDump) tracking / scheduling / automation system capable of managing a tracking station, both in a GUI-oriented fashion and ingestor-like way.

As far as 1 goes, I believe that can be solved by a similar interface to the CLI recorder with some improvements, added in the main satdump CLI interface, with perhaps similar integration to start it in the UI.

For 2, it would not too hard to implement but as I have explained in #38, there doesn't seem to be something really ideal.
GPredict utilizes hamlib, but rigctl lacks features to simply "get a NORAD ID and frequency" or similar to easily start doing the right thing on the SatDump end of things.
Orbitron (and I believe UltraTrack, but it's not software I am familiar with), utilizes DDE which does have the features required and already are used to scheduled LRPT passes by many. My main issue there is honestly that I was unable to find any proper documentation about this. (though, it doesn't solve the GPredict issue)

For 3, it's mostly due to my own issues using GPredict and other tracking software that sometimes lacked what I needed, and if I'm going to be writing something like that, I'm already halfway through having a full UI & system for managing passes and recording. I have yet to settle on the details and if it will even be in SatDump's core directly, but it is something I am working on for sure, as I will require it to manage my own station(s).

SV1CAL, 2022-01-05:
Hi Alan ,
Thanks for your lengthy answer !!!
Don't want to cause you more troubles :-) . Maybe what's needed to integrate into my satellite tracking software is already there.

You mentioned :
"For recording, you can already use the command-line recorder, alongside a command similar to "timeout" on Linux which will send a killsig after a set delay."

I am pretty sure I am missing something, but I haven't figured out how to launch the recorder from the command line.
The only applications I see in the extracted folder are satdump.exe , satdump-ingestor.exe , satdump-ui.exe, volk_profile.exe.
I haven't looked at the source code to see if the recorder or live processing could be started by just calling satdump.exe with the right arguments. That would be awesome and good enough to get started :-)

Regards,
Michael

Aang23, 2022-09-16:
@SV1CAL Added, see https://www.satdump.org/posts/rework-release/

closed 2022-09-16 by Aang23