SatDump/SatDump #83: (C)GK-2A Decryption keys could not be loaded

BH6AZP, 2021-12-24:
issue:  (C) GK-2A Decryption keys could not be loaded!!! Nothing apart from encrypted data will be decoded.

Hello, when I use offline processing mode to decode GK-2A LRIT pictures, this issue happened. So how could I add the Decryption keys to SatDump? Thank you!

Aang23, 2022-01-05:
Hi!
I need to document this better, but it mostly down to those key not being something that can really be shared.

You should follow @sam210723's guide at https://vksdr.com/xrit-rx and put the EncryptionKeyMessage.bin file (with this filename) in resources/gk2a/.

sam210723, 2022-01-05:
Here's a link to that specific part of the guide
https://vksdr.com/xrit-rx#keys

BH6AZP, 2022-01-12:
> Hi! I need to document this better, but it mostly down to those key not being something that can really be shared.
> 
> You should follow @sam210723's guide at https://vksdr.com/xrit-rx and put the EncryptionKeyMessage.bin file (with this filename) in resources/gk2a/.

Thank u very much. it works!

Aang23, 2022-01-27:
About this :
It may not be a bad idea to simplify the process.

On Linux system where SatDump was installed, it may not be possible to add something to resources. In that case it should fallback to .config/satdump/resources in my opinion (which is also planned for TLE management and other dynamic resources).

From there, perhaps some of it could be automated with less user input than it does currently... Might want to think about it sometime even if this isn't a priority at all.

sam210723, 2022-02-02:
Could be a simple `wget` to `nmsc.kma.go.kr` then unzip and decrypt in the background, which still avoids the software actually shipping with the keys

Aang23, 2023-02-22:
GK-2A keys are now automatically downloaded if not already present. 
https://github.com/SatDump/SatDump/blob/master/plugins/gk2a_support/gk2a/module_gk2a_lrit_data_decoder.cpp#L82

@sam210723 I ended up basically just downloading & extracting the file from archive.org and decrypting it (ported over your Python code). That way, doesn't ship with the keys but will be a lot easier for users.