SatDump/SatDump #84: Auto-builds

berryempire, 2022-01-06:
Github Actions can be used to auto-build binaries like .exe .deb, etc.
Please utilize it. Thanks!

ZbychuButItWasTaken, 2022-01-06:
Hi
We're already using Github Actions to generate Windows and Android builds.

Aang23, 2022-01-06:
Indeed, it has been done for a while.

.deb files are not currently generated mostly because I didn't get around to setting up CPack just yet. Otherwise, I would generate .debs for common distributions.

What were you thinking about exactly? If it's .exe, then this issue can closed, otherwise, .deb is valid and I'd reopen it.

berryempire, 2022-01-07:
Hello, I'm using Ubuntu and auto-built .deb binaries would make the process of installation much simpler.
Thanks!

berryempire, 2022-01-07:
@Aang23 Ubuntu 21.1

AlexandreRouma, 2022-01-07:
@Aang23 I can PR the same system I use for SDR++ if you want. (for the RPi builds you'll need a pi configured to as a github actions worker)

berryempire, 2022-01-26:
@Aang23 @ZbychuButItWasTaken Would it be possible, please?

ZbychuButItWasTaken, 2022-01-26:
Sure, we can put it in the TODO

closed 2022-04-19 by ZbychuButItWasTaken