SatDump/SatDump #88: Bug: Program crashes when PC goes standby

Anjum9694, 2022-01-18:
### Expected Behavior
Satdump should continue live decoding as usual.

### Current Behavior
When the PC goes standby (screen turning off, not sleep) the program either crashes or just exists itself. I have repeated the scenario on a different computer to be sure.

### Possible Solution
Implement a wake lock when Satdump is doing live decoding.

Aang23, 2022-01-20:
That's an interesting one... What OS / hardware was this tested on? Do you also happen to have any logs?

On my linux machine, I do not think I have observed that issue the few times I did put it to sleep while having some pipeline running, but I will definitely try to reproduce tomorrow.

Thanks!

Aang23, 2022-01-20:
Though to be fair, just thinking about it in the first place having any sort of SDR being utilized and then putting the machine to sleep is quite likely to cause an irrecoverable state in the live-processing pipeline.

When it's about network sources, I'd also expect the server to have kicked the satdump client out much earlier as well...

That means your proposed wake lock solution would probably be more viable as it is the user's role not to causes things SW can't handle in the first place.

Anyway, will need some testing first.

Anjum9694, 2022-01-20:
Windows, source RTL-SDR dongle.

It happens if the screen goes off when its decoding a live image, I'll post the cmd window log in the next transmission.

Anjum9694, 2022-01-23:
I just changed the screen off timer instead so that solves it. Wouldn't really consider this as a bug.

closed 2022-01-23 by Anjum9694

AlexandreRouma, 2022-01-23:
tbh this is probably a graphics driver bug that crashes opengl when it powers down the screen

Aang23, 2022-01-23:
> tbh this is probably a graphics driver bug that crashes opengl when it powers down the screen

Yes it is... Just tried in my VM and on a physical machine. Happens on Windows only.

So, nothing I can do about it on my end, and I guess that can indeed be considered a non-issue. 

Though, perhaps a software renderer wouldn't have that issue. If it's really important in your case @Anjum9694, perhaps try https://fdossena.com/?p=mesa/index.frag.