SatDump/SatDump #90: bug: gvar_extended text missing from preview.png

errolt, 2022-01-26:
It looks like, after the move to the new image lib, the text went missing from the preview.png image.
Looking at the code, the text is drawn on uninitialized image objects.
I don't know if the old image lib automatically expanded the image, but the new image lib checks that the pixels are within image bounds, and exists if not.

Images for text overlay created here:
https://github.com/altillimity/SatDump/blob/093bba360ceb3b6de65062f87ce6a7993929c872/plugins/gvar_extended/main.cpp#L122

Text written to overlay images here:
https://github.com/altillimity/SatDump/blob/093bba360ceb3b6de65062f87ce6a7993929c872/plugins/gvar_extended/main.cpp#L136-L138

draw_text uses draw_pixel which does bounds checking here:
https://github.com/altillimity/SatDump/blob/093bba360ceb3b6de65062f87ce6a7993929c872/src-core/common/image/image_drawing.cpp#L14

The result is images that are still empty, even after the text was "drawn".

ZbychuButItWasTaken, 2022-01-26:
That's very possible. Gonna be honest here: I haven't properly tested it as my station was down so I didn't have any recent data. It indeed doesn't init the images before writing. That's an easy fix though, so gonna work on it in a free moment.

errolt, 2022-01-26:
I did try an easy fix by initing all the images to (bar_height,previewImage.width()) but that had unexpected results, as the code that prints these images onto the final image expected the overlay images to only be as wide as the text in them, so one would have to init the images to the size of the text.

ZbychuButItWasTaken, 2022-01-26:
Yeah, that was the easiest way to do it in CImg. I'll change up the code a bit and it will work just fine

errolt, 2022-01-26:
BTW, no rush. I only got plugins running last night, so this isn't disrupting or urgent. Just thought I'd file a bug for it...

ZbychuButItWasTaken, 2022-01-26:
Alright! and thanks for letting me know, I wouldn't have seen it myself until I put my station back up which is gonna be in a few days...

ZbychuButItWasTaken, 2022-01-27:
Update:
I've been tinkering with it for a while and it's almost ready! I've ended up modifying the image lib, but the current font doesn't work very well in low resolution. I'll try some other fonts, but the basic mechanism works

[image]

errolt, 2022-01-27:
Great! Thanks. I also saw the weird aliasing. I'll take a look at the font scaling tonight, see if I can improve it.

ZbychuButItWasTaken, 2022-01-27:
well, it's using bilinear resizing for it. This font just doesn't look very good when resized.. Roboto mono will do better here, it's a lot "thicker"

ZbychuButItWasTaken, 2022-01-27:
(the font system, like the whole image lib is still very basic, so some things aren't done "properly" yet)

ZbychuButItWasTaken, 2022-01-28:
Just pushed a closing fix, it all works now.
[image]

closed 2022-01-28 by ZbychuButItWasTaken

errolt, 2022-01-28:
Nice! Thanks!

ZbychuButItWasTaken, 2022-01-28:
let me know if it all works well for you though

errolt, 2022-01-29:
Works great! thanks!
[image]

commit 34a57d0e (2022-01-28): Fixed the gvar_extended plugin, close #90