SatDump/SatDump #91: Processing recorded MetOp baseband files

SV1CAL, 2022-01-29:
Hi Alan,
I noticed that even though live decoding MetOp works fine, when a baseband recording is demodulated in SatDump is always stays at "nosync" and cadu files is empty. I have tried recording in HDSDR with both Lime and Airspy . I also tried recording in SDR# with no success. IQ swap, DC block etc don't make a difference.
Then I though why not try the  "old" standalone demodulator !
The resulting soft bits file was passed to SatDump and all worked fine! So the issue must be in the the demod code, I guess...

I have kept the baseband file and all the output files for the cases above and I am uploading them to my google drive.

Now one could ask why record when you can live-decode. Well, I like automatically recording passes with my own software and post-processing them later. Works great with X band but came across this on L band.
Will post the link to the files later when uploading has finished.

Thanks!
Michael

SV1CAL, 2022-01-29:
Here is the link to the w8,6MSPS file:
https://drive.google.com/file/d/1VUHVMvxuBmVb1jgorFs5QRDyns4qW-Ch/view?usp=sharing
As mentioned demod is ok with the standalone  demodulator but not with SatDump's built-in (tried earlier versions too).
Constellation / SNR looks fine but it stays at "nosync" state.

Regards,
Michael

Aang23, 2022-02-04:
That is very weird... It definitely works here on my end with any baseband file I can throw at it, and does for many other people as well.
I will try with your file and perhaps check if it isn't a weird windows edge-case or so.

Aang23, 2022-02-04:
Ok well, that was simple.... Just made an error in the code reading HDSDR-style files...

closed 2022-02-04 by Aang23