SatDump/SatDump #94: Change GOES HRIT backgrounds to black

JB-Micro, 2022-02-16:
Is there a way to make live decoded images be saved with a black background instead of a white background?

zeke800, 2022-02-16:
I would like this feature too. I just select GOES-R HRIT to goestools and then process to goestools.

Aang23, 2022-02-17:
Yes, that could surely be done.
The reason the background is white simply is that it's what it is in the raw imagery sent by NOAA.

JB-Micro, 2022-09-10:
I've been thinking about this more recently. I hadn't realized that the borders slightly change with each image so having a single background template won't fit all images perfectly. The fix is to make a new background mask from the current channel 2.

This change would require an edit to the line I previously posted. 
```
for (int c = 0; c < 3; c++){
    falsecolor[c * falsecolor.width() * falsecolor.height() + i] = fc_lut[c * fc_lut.width() * fc_lut.height() + x * fc_lut.width() + y] - (ch2[i]-254)*255;  

    if(falsecolor[c * falsecolor.width() * falsecolor.height() + i] < 0)
        falsecolor[c * falsecolor.width() * falsecolor.height() + i]=0;
}
```
This takes the channel 2 image and does a basic threshold function on it, turning all values less than 255 to 0, and all values that are exactly 255 stay 255. This will make the border perfect each time, and I haven't seen a channel 2 image where there are 255 values inside the Earth's borders so I suspect that won't be an issue.

I'm running Satdump on windows and haven't been able to try compiling this to see if it works the way I expect and I haven't touched C++ in years, but when I tried similar code in Matlab with  channel 2 and channel 13 images from satdump, the result was perfect.

ZbychuButItWasTaken, 2022-09-16:
that can be achieved through the composite system now, see newest release. You gotta enable the automatic composite generation in settings and use a ternary expression to replace all 255 values with a 0

Aang23, 2022-09-16:
Actually no, it can't @ZbychuButItWasTaken. Well it can, but not reliably. So I'd still consider this open and to be handled when saving products for both HRIT and GRB.

JB-Micro, 2022-09-25:
You're right, that solution can still break down. Oddly enough, it seems to be the way goestools does its backgrounds and I haven't seen any stray black pixels. Anyways, my new solution is a bit more involved. I take channel 2 and scan it line by line in all 4 directions, and when I first encounter a non-white pixel I set it and all the proceeding pixels in that scan to black to make a piece of a mask. Ignore the inconsistent screenshotting skills below:

[image]
[image]
[image]
[image]

 Then I do a logical AND on each pair of adjacent mask pieces, which gives me 4 white quadrants. The logic here is that all the pixels I want to turn black will always "see" at least 2 image borders and most if not all of the 255-valued pixels that are part of the earth image will only see 1 border or none. Here's the upper left quadrant:
[image]

 I can finally add those together to get the full mask. This won't turn any internal 255-valued pixels to black, but it will turn those pixels black if they're on the very edge of the Earth. I think that in the worst case scenario, this only incorrectly converts a sliver of pixels that is nearly unnoticeable. I've gotten this to work in matlab and am going to implement it in python. I'm not even going to attempt it in c++. I think the risk and impact of errors is low enough in this solution that I'm not going to keep refining it. Maybe you guys can think of a better solution though.

JVital2013, 2023-07-22:
I'm not sure what the plans are for this feature request, but I took a stab at a fix because I've had too much time on my hands and access to a C++ compiler 😄 

I recently ported Goestools to Windows and looked at how it solves this problem. It doesn't _quite_ just replace all 255s with 0s - what it does is much more clever IMO. It starts by looking at the very left and right pixel on a given line, and turns them from 255 to 0 if necessary. It then works its way into the middle of the line, and stops as soon as it finds a pixel that is not 255. It then moves on to the next line - rinse and repeat until the entire image is taken care of. This prevents any stray black pixels in the middle of the image, and does a pretty good job of changing the background color. The only real risk is that a true white pixel on the very edge of the disk may be turned black, but the chance of this is low. It would have been indistinguishable from the background anyway.

Now, we don't want to be a copy goestools, but I re-implemented something similar into SatDump. It's not a copy and paste reimplementation - I did not reference goestools, other than what was still floating around in my noggin 😄. Also, my implementation allows changing any background color to any other color, and should be fairly easy to implement into other pipelines with Full Disk imagery.

**If you want to check it out, I have the changes [here](https://github.com/JVital2013/SatDump/commit/fa477441f377edf34e9aa08c8777c79ac3afedac) and you can download builds [here](https://github.com/JVital2013/SatDump/actions/runs/5628402793).** Might be of interest to you @JB-Micro and @zeke800.

Right now, this implementation only applies the background correction to Channel 2 segments as they come in, which fixes it for Channel 2 and False Color images. People might want something similar for the GOES Himiwari relay, but I don't have access to GOES-18 HRIT to test how it works. **Anyone got a CADU?**

**All that said: I did not open a PR for this because I'm not sure if it's how we want to proceed:**
- I don't want to step on @ZbychuButItWasTaken or @Aang23's toes if they have another fix in the works
- Should background correction always be on, or should it be a pipeline option? I think most people would want the backgrounds to be corrected to black, but there's always someone who might not want this.

Thoughts? Here's an example image from a version of SatDump with background color correction:

[image]

ZbychuButItWasTaken, 2023-07-22:
While I am not near a computer for a while now (so I can't check it out), sounds great to me! I'll have a look at it when I can (Next month) or Aang maybe can earlier.

Anyway, sounds like it should work nicely and we definitely will look into implementing that! Thanks for the suggestion, and I think a PR would be siutable here (we don't have a solution as of now afaik, but Iam slightly under a rock recently)

JVital2013, 2023-07-22:
I went ahead and submitted a PR, then you two can look at it whenever you get the chance! Let me know if you want me to adjust anything

JB-Micro, 2023-07-22:
I'll try it out. That solution from goestools sounds functionally identical to what I did, except they skipped the ANDing part. I doubt that would change too much, maybe a pixel here or there every few days.

closed 2024-04-30 by Aang23