SatDump/SatDump #95: no longer updates the TLEs, from Satdump update TLE

IZ5RZR, 2022-02-16:
Hello Altillimity
Thanks for this great software
For a couple of days the program no longer updates the TLEs, while through ORBITRON I noticed that the TLEs are downloaded from celestrack.com.
Is it possible that some links have changed?, can you fix it?
Hello
IZ5RZR

Aang23, 2022-02-25:
Do you have the logs on hand? It's been fine on my end.

Robydem, 2022-02-28:
Hello, same problem, it does not update the data from celestrack.com. tnx

IZ5RZR, 2022-02-28:
> Do you have the logs on hand? It's been fine on my end.

Actually not, but problem still persist.
You can try it windows version 0039, tle wont download
erros is can't fetch tle data.
The only way to update manually is edit old tle from resources\tle folder and search number of satellite in celestrack, copy tle data and paste in correct file in TLE satdump folder

IZ3ZLU, 2022-03-02:
[image]

IZ3ZLU, 2022-03-02:
> Mah, questo non risponde.
>
> Non è che l'hanno mandato al fronte i russi?
>
> Da "IZ3ZLU" ***@***.***
> A "altillimity/SatDump" ***@***.***
> Cc "Max" ***@***.***,"Author" ***@***.***
> Data Tue, 01 Mar 2022 22:01:44 -0800
> Oggetto Re: [altillimity/SatDump] no longer updates the TLEs, from Satdump
> update TLE (Issue #95)

Spero di no..... poveretto.....

IZ5RZR, 2022-03-26:
> Hello, same problem, it does not update the data from celestrack.com. tnx

Bug still continue. No TLE update with windows systems

x31415926, 2022-04-10:
It seems the configured TLE address (from json) returns HTTP moved permanently. The new location, as far as I can see is https://www.celestrak.com/NORAD/elements/gp.php?CATNR=

As an aside, when/if someone fixes this, might as well fix this line as well since the formatting for spdlog doesn't use the C formatters. 
logger->error("HTTP Server Responded: %d %s", nng_http_res_get_status(res), nng_http_res_get_reason(res));

IZ5RZR, 2022-04-21:
Hello! Fix working good by copy TLE.Json from repository to  Satdump resources folder.
Instead I have make a simply batch script for windows 10 to update tle to run in satdump resources tle folder
[run - SatdumpTLE.zip](https://github.com/altillimity/SatDump/files/8529570/run.-.SatdumpTLE.zip)

closed 2022-09-16 by ZbychuButItWasTaken