SatDump/SatDump #97: HackRF One not recognised by SatDump

pluggerlockett, 2022-02-21:
I've got a Ali HackRF One clone, works fine in Gqrx. SDR++ works with it once I remove the soapy drivers as documented on that project's Github page. But I can't get SatDump (v0.38 or v0.39) to recognise the device. I've hit the refresh button, unplugged completely, rebooted etc. I've updated the device firmware from 2018 to 2021.03.1 and still no luck. I'm on Debian 10 as my base OS.

Aang23, 2022-03-05:
That's weird, especially that it works fine with SDR++.

Did you have libhackrf-dev installed at the time of building? Perhaps HackRF support was not enabled at build time?

pluggerlockett, 2022-03-07:
Hi Aang23. Apologies, you were correct. I didn't have libhackrf-dev installed. Installed it, reinstalled SatDump, and now the HackRF is recognised fine. Thanks for pointing out my mistake, I appreciate it.

closed 2022-03-07 by pluggerlockett