SatDump/SatDump #98: NOAA/METEOR HRPT not syncing correctly

Crosswalkersam, 2022-02-23:
Hello,

I tried receiving NOAA / METEOR HRPT with an Airspy Mini @ 3MSPS and Satdump 0.0.39. Its Win 10 with an i5 processor and 16GB of RAM.
The satellite is there and clearly visible in the spectrum when I record a baseband. 
When I try to demodulate it however, I get something like this:
[image]
The BPSK is clearly splitting into two groups, but connected with two "bridges", and the Deframer does not sync.
However: After 2/3 Minutes, it suddenly starts working:
[image]
You can see that the SNR jumps from 6dB to 9/10dB almost instantly, and the deframer syncs fine.
I tried enabling DC-Block, but that did not fix the issue. This happens with NOAA and METEOR.
Decoding with a RTL-SDR@2.56MSPS, I never had that happen. If you need, I can supply the .i16 file recorded with satDump aswell.
Thanks!

Aang23, 2022-03-05:
That's weird. Did DC-Block help? Otherwise, yes it will be easier to just look at the .i16 on my end.

Crosswalkersam, 2022-03-06:
DC-Block does not help. [Here](https://mega.nz/folder/yiZlnSqT#F3EfTt5V0v_oyY4A5gUP9g) is a folder that contains a recording of NOAA18 at 3MSPS.

Crosswalkersam, 2022-03-31:
I just checked, the issue goes away when sampling at 6MSPS. I guess the Airspy behaves a bit weird at 3MSPS?

zeke800, 2022-04-02:
Apparently, [yes](https://www.reddit.com/r/amateursatellites/comments/rzo0k8/problems_with_my_new_hrpt_setup/).

closed 2022-04-19 by ZbychuButItWasTaken

mapaknot, 2022-04-28:
Hey
Synchronization is interrupted. I changed one setting and everything was fine for me.
#####"rrc_alpha": 0.5, #####

    "frequencies": [
               1707
               1698
               1702.5
           ],
           "sampling rate": 3e6,
           "baseband_type": "i16",
           Work: {
               "main band": {},
               soft: {
                   "pm_demod": {
                       "symbolrate": 665400,
                       "agc_rate": 0.002e-3,
                       pll_bw: 0.01
                       pll_max_offset: 0.5
                       "rrc_alpha": 0.5, 
                       "rrc_taps": 31,
                       "costas_bw": 0.005,
                       "clock_omega_relative_limit": 0.005,
                       "buffer_size": 8192

> Здравствуйте,
> 
> Я попытался получить NOAA / METEOR HRPT с Airspy Mini @ 3MSPS и Satdump 0.0.39. Его Win 10 с процессором i5 и 16 ГБ оперативной памяти. Спутник есть и хорошо виден в спектре, когда я записываю базовую полосу.Однако, когда я пытаюсь демодулировать его, я получаю что-то вроде этого: [image] BPSK явно разделяется на две группы, но соединен двумя "мостами", и Дефрамер не синхронизируется. Однако: через 2/3 минуты он внезапно начинает работать: [image] Вы можете видеть, что SNR прыгает с 6 дБ до 9/10 дБ почти мгновенно, а дефрамер синхронизируется нормально. Я попытался включить DC-Block, но это не устранило проблему. Это происходит с NOAA и METEOR. Decoding с [RTL-SDR@2.56MSPS](mailto:RTL-SDR@2.56MSPS)- Со мной такого никогда не случалось. Если вам нужно, я могу предоставить файл .i16, записанный с помощью satDump.Спасибо!

mapaknot, 2022-04-28:
Hey
Synchronization is interrupted. I changed one setting and everything was fine for me.
#####"rrc_alpha": 0.5, #####

    "frequencies": [
               1707
               1698
               1702.5
           ],
           "sampling rate": 3e6,
           "baseband_type": "i16",
           Work: {
               "main band": {},
               soft: {
                   "pm_demod": {
                       "symbolrate": 665400,
                       "agc_rate": 0.002e-3,
                       pll_bw: 0.01
                       pll_max_offset: 0.5
                       "rrc_alpha": 0.5, 
                       "rrc_taps": 31,
                       "costas_bw": 0.005,
                       "clock_omega_relative_limit": 0.005,
                       "buffer_size": 8192

> Hello,
> 
> I tried receiving NOAA / METEOR HRPT with an Airspy Mini @ 3MSPS and Satdump 0.0.39. Its Win 10 with an i5 processor and 16GB of RAM. The satellite is there and clearly visible in the spectrum when I record a baseband. When I try to demodulate it however, I get something like this: [image] The BPSK is clearly splitting into two groups, but connected with two "bridges", and the Deframer does not sync. However: After 2/3 Minutes, it suddenly starts working: [image] You can see that the SNR jumps from 6dB to 9/10dB almost instantly, and the deframer syncs fine. I tried enabling DC-Block, but that did not fix the issue. This happens with NOAA and METEOR. Decoding with a [RTL-SDR@2.56MSPS](mailto:RTL-SDR@2.56MSPS), I never had that happen. If you need, I can supply the .i16 file recorded with satDump aswell. Thanks!