SatDump/SatDump #100: projection wrong on S-N pass

marianzb, 2022-02-25:
projected images from polar satelite are wrong during northward pass (nightly). This is from Meteor M2 LRPT, daytime is ok, nighttime has wrong position and shape.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
.

Aang23, 2022-02-25:
To make sure : were all those projections done with data processed on the (UTC) day of reception? They have to be as M2 does not transmit any sort of day indicator.

This looks more like a LRPT-specific issue. It is known to work with the rest.

K4KDR, 2022-02-25:
I see the same issue "sometimes" with S-to-N (that is nighttime here in U.S.) passes.  However, at times the projection is correct.  I am only using Meteor-M2 LRPT 'live processing'.   

I have never noticed if the time was past 0000 UTC - will watch that going forward to see if the "good" projections are before 0000 UTC and the reversed projections are after 0000 UTC or not.

marianzb, 2022-02-25:
Thanks for this hint, its caused by processing on next day. Now I have processed it same day as it was recorded and its correct.
I would have a suggestion if M2 decoded time is in 'future' according current UTC time, there could be subtracted one day. It will solve this problem at least for case when processing and recording were within 24 hours.

[image]
[image]

K4KDR, 2022-02-25:
Actually, I see now that my issue is **not** related to time-of-day.

Both the live capture & the final processing were done before 0000 UTC.   If you would like this to move to a separate issue, please let me know.

As is normal with south-to-north passes, the Meteor-M2 LRPT image is inverted.  However, the green border outline is not.

The result is that the image is correct (inverted) but the green borders are not.  As I mentioned previously, about one time out of every 10, the orientations will match.  I do not know what is different about the times that they match correctly.

This image should render as being over North America.

[image]

Aang23, 2022-02-26:
Wait. I forgot - M2 is not in UTC time but Moscow time, unlike M2-2.
Would that explain your issues?

K4KDR, 2022-02-26:
I believe you are correct.  I am now on the same UTC day as Moscow time (26-Feb) and I just ran last night's .CADU file through the 'Offline Processing' tab.  See below - the projection is correct.     Other than changing my **system time zone** to match Moscow time, do you have any suggestion for how to render these correctly when using 'Live Processing'?  Thanks!

[image]

K4KDR, 2022-02-26:
Did a test on tonight's Meteor-M2 pass to see if setting my system time/date zone to Moscow would result in the border overlay being correct.  It did not.

Before the SatDump-UI app was run, my system timezone was set to Moscow (shown here at the end of processing, but it was set prior to running the application):

[image]

... but the end result after "Live Processing' was that the border overlay was reversed:

[image]

... I noticed something that I cannot explain.  Even though my system time, as well as the time-stamps that run down the left side of the terminal window that SatDump was run from, all show Moscow time, the post-pass processing run by SatDump created its folder structure using UTC time:

[image]

... so, I hope that this is helpful to perhaps explain the day mismatch from Moscow time & the resulting inverted border overlay on these late night south-to-north passes.  Many thanks!

closed 2022-04-19 by ZbychuButItWasTaken