SatDump/SatDump #99: multiple batch decoding no result

marianzb, 2022-02-25:
I have noticed this problem on earlier versions, but in #465 its real problem. I have batch file to offline decode recorded files (VHF, Meteor M2). Using satdump-ui.exe no problem, those are decoded each, one by one manually, but batch file processing fails to decode, created no files and .cadu have zero size. It looks like it needs that ui window to process it properly. You can test it with batch file decode.bat containing something like this:

satdump.exe meteor_m2_lrpt baseband meteor_2202250621_137102000Hz.u8.wav products out_satdump\meteor_2202250621_137102000Hz.u8 -samplerate 240000 -baseband_format w8
satdump.exe meteor_m2_lrpt baseband meteor_2202250443_137102000Hz.u8.wav products out_satdump\meteor_2202250443_137102000Hz.u8 -samplerate 240000 -baseband_format w8

when satdump.exe is replaced by satdump-ui.exe everyting is correctly decoded but windows needs to be manually closed to continue batch

Aang23, 2022-02-25:
Hmmmmm.... That sounds like a windows-only issue to me.
Do you have logs on hand?

marianzb, 2022-02-26:
run both versions with satdump.exe and satdump-ui.exe, here are the logs, I hope it will help to find the problem.
[log1.txt](https://github.com/altillimity/SatDump/files/8146039/log1.txt)
[log2.txt](https://github.com/altillimity/SatDump/files/8146040/log2.txt)

Aang23, 2022-03-05:
Would you be able to share the files this happened with? I will try to reproduce on my end.

marianzb, 2022-03-05:
For example this file, its 8bit u8 (rtl_sdr). But looks its not file dependent, sometimes it just don't decode without ui, sometimes yes, try more times with or without ui. Here is the link (1 week expity) https://we.tl/t-GnPIZJJDCc

Aang23, 2022-09-16:
Would you be able to check if this is an issue now?

marianzb, 2022-09-17:
unfortunately i have damaged antenna now, i can try it later when some data will be received

ZbychuButItWasTaken, 2022-09-26:
Well, I will close this for now. If it doesn't work when you hsve the data feel free to open

closed 2022-09-26 by ZbychuButItWasTaken