SatDump/SatDump #104: A couple issues with SatDump v0.39 and a Wiki typo

pluggerlockett, 2022-03-29:
Hi Alan,

Just wanted to let you know I've encountered a couple issues with Satdump v0.39. I'm running it on Debian 10. Both of the issues relate to GEO "Live processing" pipelines. Specifically:

GK-2A LRIT to xrit-rx Pipeline doesn't connect successfully to xrit-rx. SatDump v0.38 does work though. I've removed and reinstalled xrit-rx to see if an update over there would fix it, but the issue persists.

The Elektro-L LRIT Pipeline live image decoding doesn't work on my machine. Once live processing is complete (from an Elektro-L transmission perspective) and live processing is stopped, the recorded cadu can then be processed successfully by the "Offline processing" plugin.

And finally, on the Ingestor Wiki page the RTL-SDR template is missing double quotes around the AGC value, which causes the ingestor to crash. The line should read `"agc": "0"`

pluggerlockett, 2022-04-14:
Just wanted to provide a quick update on the Elektro-L LRIT live processing (but not live decoding) issue I'm seeing. Live processing works fine on Win10 and doesn't leave a cadu file, instead it just processes and outputs the images. Conversely on Debian 10 live processing does not output the images, instead it just leaves a cadu file. It appears to be Linux specific, and I've confirmed it doesn't work via the satdump-ui or the satdump-ingestor on two separate hosts.

sdrgeek, 2022-07-27:
It works on my machine eunning Debian 11.

pluggerlockett, 2022-07-28:
I might give it another whirl, but 2x clean builds of Debian 11 (one x64, one ARM) back in March didn't work for me.

sdrgeek, 2022-07-28:
Elektro-L N3 processing works perfectly fine on Debian 11, but I'm not able to test GK-2A since it's not in range. If not already done, try building from master.
What's wrong with using your Airspy on your Laptop or pi?

pluggerlockett, 2022-07-28:
Ah, on Debian 10 (ARM or x64) live processing of L3 in v0.39 doesn't work and just outputs a .cadu that I offline process after the fact. On Win10 it works without issue.
Nothing's wrong with my hardware, it's just that there's an issue with the newer libusb update that's coupled with Debian 11. My RTLSDR v3 works fine on Debian 11 (ARM or x64) but the airspy crashes. See Issue [#106](https://github.com/altillimity/SatDump/issues/106) in the Closed section.
Which kinda forces me to stay on Debian 10 as I get better SNR with my Airspy, hence its inclusion in my L3 station instead of a RTLSDR.

pluggerlockett, 2022-07-28:
at the time it occurred in Satdump UI and the ingestor. Maybe some underlying issue got resolved. I'll try again in the near future, buster's getting a bit long in the tooth.

closed 2022-09-16 by ZbychuButItWasTaken