SatDump/SatDump #106: Airspy doesn't work with satdump-ingestor on Raspberry Pi

pluggerlockett, 2022-04-11:
Satdump-Ingestor works fine on Debian 10 (SatDump v0.39) with a Airspy mini.
When attempting to run the ingestor on a Raspberry Pi with an identical configuration, it crashes with a "Could not find requested SDR Device!" despite it appearing as connected via lsusb. See below. I've tried on two separate Pi's, both running Satdump v0.39. I've enabled and disabled Bias T in the config.json file without success. And I've tested with a RTL-SDR v3 instead of the Airspy Mini, and that works fine.

lsusb output
[image]

Satdump-ingestor error
[image]

pluggerlockett, 2022-04-14:
Hi Alan, just wanted to update here that I don't believe this is an issue with SatDump but with Debian 11. I built a NUC with Debian 11 to use in the station to test yesterday as I assumed the Airspy error was related to ARM/Rasbian. Come to find out the NUC running Debian 11 x64 spat the same error. So I rebuilt with Debian 10 and now the Airspy is working and I'm getting SYNC. :D
I'll rebuild one of my Raspberry Pi's with Debian 10 and see if I can get it to work and SYNC with my Airspy mini. But in the interim I'd recommend folks steer clear of Debian 11 if you want to use SatDump with an Airspy.

Aang23, 2022-04-21:
Unfortunately, you made the right observation.

As @AlexandreRouma  told me when that came out, a recent libusb update broke most SDR libraries, including libairspy. Debian 11 ships this new version, which hence does not work properly.

I would recommend pinning it to an earlier version for now, nothing else I can do about it on my end.

pluggerlockett, 2023-02-09:
Hey Aang and Zbychu. Just a quick heads up; I think I found the root cause of this issue. It was an unmet dependency. As mentioned in this thread and the other linked above I was having issues with Debian 11 and SatDump, both on arm and x64. I rebuilt a new Debian 11 Rasbian box to put SatDump 1.0.3 on and the issue persisted. So I did the same on a Nuc on x64, it also would not work. So I went and installed the SDR++ dependencies on the Nuc and it's been ingesting FY-2G svissr data ever since. I just swapped out my Elektro-L3 station Debian 10 Pi for a Debian 11 one, and that's running fine headless as well after installing the SDR++ prerequisites. I'm not sure exactly _which_ package fixed it for me, but maybe some of the Soapy Airspy stuff?
Regardless, it's working now, and combined with the Linux live processing now working (so I don't have to manually handle/process cadu files) I'm very pleased! Thanks for all y'alls hard work.