SatDump/SatDump #107: error : Could not set SDR frequency

YD9CKH, 2022-04-17:
Can someone tell me how to resolve the error. Im using Satdump v0.0.38 for Windows, When try to START Live processing for pipeline : GK-2A LRIT, i get error  " (E) Could not set SDR frequency " and the apps is stuck. 
Thank you,
[image]

zeke800, 2022-04-18:
What SDR are you using?

YD9CKH, 2022-04-19:
Hi,
Im using SDR V3 and my Windows is Windows 10 64bit

Thank you
Best Regards,

YD9CKH

zeke800, 2022-04-20:
Hmm you mean the RTL SDR V3? I have one of those and it works quite well. Try running rtl-test and see if it works. Occasionally there have been faulty dongles. Does it work with SDR software such as SDR# or SDR++ ?

closed 2022-09-16 by ZbychuButItWasTaken