SatDump/SatDump #110: Meteor MN2 channels align

marianzb, 2022-04-28:
Sometimes, quite often, there is problem with composing color picture from decoded channels. Original channels are decoded correcly, but 125 or 123 composite is wrong
[image]
[image]
[image]
[image]

here is attached also signal source, thanks
[cadu_source.zip](https://github.com/altillimity/SatDump/files/8580581/cadu_source.zip)
.

marianzb, 2022-04-28:
another example
[image]
[image]
[cadu_source2.zip](https://github.com/altillimity/SatDump/files/8580650/cadu_source2.zip)

zeke800, 2022-05-25:
Hmm... just looking at it it seems correct.

closed 2022-09-16 by ZbychuButItWasTaken