SatDump/SatDump #112: Error with HackRF

IZ3ZLU, 2022-05-10:
I first connected my HackRF-One to SatDump today.
It is seen regularly but when I start receiving the message "Could not open HackRF device!"
In reality everything seems to work, even if I haven't tried to receive yet ..
Thanks for any info
Lanfranco IZ3ZLU
[image]

IZ3ZLU, 2022-05-11:
I use Windows 10...... excuse me.
Many thanks.

IZ3ZLU, 2022-05-11:
I solved!!!!
I downloaded the latest version of Zadig which installed the 10.xxxx drivers instead of the 9.xxxx ones. Now the error message has disappeared.
Many thanks

IZ3ZLU, 2022-05-11:
Now in Live Processing everything works .... but in Baseband Recorder it shows the error again ....
many thanks

ZbychuButItWasTaken, 2022-05-11:
As you said, it does not change anything and pricessing works just fine. It comes from the driver, we can't really do anything about it.

IZ3ZLU, 2022-05-11:
Many thanks.