SatDump/SatDump #114: Windows builds

zeke800, 2022-06-04:
Would it be possible to make the windows builds work again? The older ones are expired, and the release is old. It is now impossible to download the new version because for some reason it can't download SDRPlay.zip. However, the android builds work well. 

Run Invoke-WebRequest -Uri "https://files.altillimity.com/SDRPlay.zip" -OutFile D:\a\SatDump/sdrplay.zip
Invoke-WebRequest : Unable to connect to the remote server

Invoke-WebRequest -Uri "https://files.altillimity.com/SDRPlay.zip" -O 

CategoryInfo          : InvalidOperation: (System.Net.HttpWebRequest:HttpWebRequest) [Invoke-WebRequest], WebException
FullyQualifiedErrorId : WebCmdletWebResponseException,Microsoft.PowerShell.Commands.InvokeWebRequestCommand

zeke800, 2022-06-05:
I now have a windows build here. https://github.com/zeke800/SatDump/actions/runs/2443641098

ZbychuButItWasTaken, 2022-09-17:
Windows builds are generated again

closed 2022-09-17 by ZbychuButItWasTaken