SatDump/SatDump #126: v1.0 make error on Ubuntu 18.04

K4KDR, 2022-09-16:
Thanks for the new version!

The root cause of this error is not obvious to me, so after not finding anything online to point me in the right direction, thought I would open an issue here in case anyone else is seeing the same thing.

At the 'make' command, I get:

```
[ 11%] Building CXX object src-core/CMakeFiles/satdump_core.dir/common/dsp/moving_average.cpp.o
/home/k4kdr/satdump_v1.0/src-core/common/dsp/moving_average.cpp:9:51: error: ‘void volk_32fc_x2_add_32fc(lv_32fc_t*, const lv_32fc_t*, const lv_32fc_t*, unsigned int)’ redeclared as different kind of symbol
                            unsigned int num_points)
                                                   ^
In file included from /home/k4kdr/satdump_v1.0/src-core/./common/dsp/buffer.h:4,
                 from /home/k4kdr/satdump_v1.0/src-core/common/dsp/block.h:3,
                 from /home/k4kdr/satdump_v1.0/src-core/common/dsp/moving_average.h:3,
                 from /home/k4kdr/satdump_v1.0/src-core/common/dsp/moving_average.cpp:1:
/usr/local/include/volk/volk.h:1445:36: note: previous declaration ‘void (* volk_32fc_x2_add_32fc)(lv_32fc_t*, const lv_32fc_t*, const lv_32fc_t*, unsigned int)’
 extern VOLK_API p_32fc_x2_add_32fc volk_32fc_x2_add_32fc;
                                    ^~~~~~~~~~~~~~~~~~~~~
src-core/CMakeFiles/satdump_core.dir/build.make:929: recipe for target 'src-core/CMakeFiles/satdump_core.dir/common/dsp/moving_average.cpp.o' failed
make[2]: *** [src-core/CMakeFiles/satdump_core.dir/common/dsp/moving_average.cpp.o] Error 1
CMakeFiles/Makefile2:643: recipe for target 'src-core/CMakeFiles/satdump_core.dir/all' failed
make[1]: *** [src-core/CMakeFiles/satdump_core.dir/all] Error 2
Makefile:155: recipe for target 'all' failed
make: *** [all] Error 2
```

System here is Ubuntu 18.04 & SatDump v0.38 runs fine.

Thanks!

-Scott,  K4KDR

Aang23, 2022-09-16:
Let me see @K4KDR!

Aang23, 2022-09-16:
@K4KDR should be fixed in the latest commit!

K4KDR, 2022-09-16:
Thanks SO much for checking on that.

Is perhaps the same issue present in other files?   I get what appears to be the same error on the next step in the 'make' process:

```
[ 11%] Building CXX object src-core/CMakeFiles/satdump_core.dir/common/dsp/moving_average.cpp.o
[ 11%] Building CXX object src-core/CMakeFiles/satdump_core.dir/common/dsp/pll_carrier_tracking.cpp.o
/home/k4kdr/satdump/src-core/common/dsp/moving_average.cpp:9:51: error: ‘void volk_32fc_x2_add_32fc(lv_32fc_t*, const lv_32fc_t*, const lv_32fc_t*, unsigned int)’ redeclared as different kind of symbol
                            unsigned int num_points)
                                                   ^
In file included from /home/k4kdr/satdump/src-core/./common/dsp/buffer.h:4,
                 from /home/k4kdr/satdump/src-core/common/dsp/block.h:3,
                 from /home/k4kdr/satdump/src-core/common/dsp/moving_average.h:3,
                 from /home/k4kdr/satdump/src-core/common/dsp/moving_average.cpp:1:
/usr/local/include/volk/volk.h:1445:36: note: previous declaration ‘void (* volk_32fc_x2_add_32fc)(lv_32fc_t*, const lv_32fc_t*, const lv_32fc_t*, unsigned int)’
 extern VOLK_API p_32fc_x2_add_32fc volk_32fc_x2_add_32fc;
                                    ^~~~~~~~~~~~~~~~~~~~~
src-core/CMakeFiles/satdump_core.dir/build.make:929: recipe for target 'src-core/CMakeFiles/satdump_core.dir/common/dsp/moving_average.cpp.o' failed
make[2]: *** [src-core/CMakeFiles/satdump_core.dir/common/dsp/moving_average.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:643: recipe for target 'src-core/CMakeFiles/satdump_core.dir/all' failed
make[1]: *** [src-core/CMakeFiles/satdump_core.dir/all] Error 2
Makefile:155: recipe for target 'all' failed
make: *** [all] Error 2
```

Aang23, 2022-09-16:
Interesting, I did not have the issue here. Did you do a fresh clone of the repository?

K4KDR, 2022-09-16:
Yes - deleted the previous folder entirely to be safe.

Aang23, 2022-09-16:
Can you entirely clean the build folder, run the cmake command and send the logs here? I want to see the output of one step.

K4KDR, 2022-09-16:
Deleted the build folder & after re-creating it, ran the cmake (for linux from the README).   The build folder afterwards is attached! 

[build.zip](https://github.com/altillimity/SatDump/files/9590518/build.zip)

Aang23, 2022-09-16:
I just pushed something that may help. Could you try it?

K4KDR, 2022-09-16:
The same, I think  (after deleting build folder & 'git pull -v').   Here is a color view:

[image]

Aang23, 2022-09-16:
Could you show the output of the cmake [...] .. command? (On first run)

K4KDR, 2022-09-16:
Absolutely!

```
k4kdr@3010i5:~/satdump$ mkdir build
k4kdr@3010i5:~/satdump$ cd build
k4kdr@3010i5:~/satdump/build$ cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr ..
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Compiling with OpenMP support!
Found SSE features, enabling building SSE code
-- Looking for CL_VERSION_2_2
-- Looking for CL_VERSION_2_2 - not found
-- Looking for CL_VERSION_2_1
-- Looking for CL_VERSION_2_1 - not found
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - not found
-- Looking for CL_VERSION_1_2
-- Looking for CL_VERSION_1_2 - not found
-- Looking for CL_VERSION_1_1
-- Looking for CL_VERSION_1_1 - not found
-- Looking for CL_VERSION_1_0
-- Looking for CL_VERSION_1_0 - not found
-- Could NOT find OpenCL (missing: OpenCL_LIBRARY OpenCL_INCLUDE_DIR) 
OpenCL NOT found or disabled!
-- Performing Test STD_HAS_FILESYSTEM
-- Performing Test STD_HAS_FILESYSTEM - Success
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Checking for module 'volk'
--   Found volk, version 2.0
-- Looking for volk/volk_alloc.h
-- Looking for volk/volk_alloc.h - not found
-- VOLK does not have volk_alloc.hh, including volk_includes
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
-- Found PNG: /usr/lib/x86_64-linux-gnu/libpng.so (found version "1.6.34") 
-- Checking for module 'fftw3f'
--   Found fftw3f, version 3.3.7
-- Performing Test VOLK_HAS_volk_32fc_x2_add_32fc
-- Performing Test VOLK_HAS_volk_32fc_x2_add_32fc - Failed
Building the GUI
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Found GLEW: /usr/include (found version "2.1.0") 
Building with Airspy support
Building with AirspyHF support
Building with HackRF support
LimeSDR Library could not be found! Not building.
SDRlay Library could not be found! Not building.
Building with PlutoSDR support
BladeRF Library could not be found! Not building.
Building with RTL-SDR support
Building with MiriSDR support
Building with SpyServer support
-- Configuring done
CMake Warning at plugins/sdr_sources/plutosdr_sdr_support/CMakeLists.txt:12 (add_library):
  Cannot generate a safe runtime search path for target plutosdr_sdr_support
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libad9361.so.0] in /usr/lib may be hidden by files in:
      /usr/local/lib

  Some of these libraries may not be found correctly.

-- Generating done
-- Build files have been written to: /home/k4kdr/satdump/build
k4kdr@3010i5:~/satdump/build$ 
```

Aang23, 2022-09-16:
That is weird. How did you install Volk?

-- Performing Test VOLK_HAS_volk_32fc_x2_add_32fc
-- Performing Test VOLK_HAS_volk_32fc_x2_add_32fc - Failed

As this fails, this issue should not be able to happen at all!

K4KDR, 2022-09-17:
As I see a 'volk' folder (repository clone) in my home directory, I assume it was installed from source long ago.   I located the binaries installed in /usr/local/bin & the version appears to be '2.0'

```
k4kdr@3010i5:/usr/local/bin$ volk-config-info --version
2.0
```

... there are also some related (I think) libraries installed from the default repository:

```
libvolk1.0-bin:
  Installed: (none)
  Candidate: (none)
  Version table:
libvolk1.0-dev:
  Installed: (none)
  Candidate: (none)
  Version table:
libvolk1-bin:
  Installed: 1.3-3
  Candidate: 1.3-3
  Version table:
 *** 1.3-3 500
        500 http://us.archive.ubuntu.com/ubuntu bionic/universe amd64 Packages
        100 /var/lib/dpkg/status
libvolk1-dev:
  Installed: 1.3-3
  Candidate: 1.3-3
  Version table:
 *** 1.3-3 500
        500 http://us.archive.ubuntu.com/ubuntu bionic/universe amd64 Packages
        100 /var/lib/dpkg/status
libvolk1.3:
  Installed: 1.3-3
  Candidate: 1.3-3
  Version table:
 *** 1.3-3 500
        500 http://us.archive.ubuntu.com/ubuntu bionic/universe amd64 Packages
        100 /var/lib/dpkg/status
libvolk2.4:
  Installed: 2.4.1-2
  Candidate: 2.4.1-2
  Version table:
 *** 2.4.1-2 100
        100 /var/lib/dpkg/status
```

Aang23, 2022-09-17:
Ok, so that is your issue, mixed-up verions! I would *highly* recommend you clean everything up and re-install a single one cleanly. Either manually, or through APT as you prefer.

K4KDR, 2022-09-17:
Thanks so much!  I will look into that.

Much appreciated!

Aang23, 2022-09-17:
You're welcome! If it was done through APT, uninstalling either 2.x or 1.x (whichever does not break other dependencies) should do the trick.

K4KDR, 2022-09-17:
Using 'apt-get remove' to uninstall all of the '**volk**' libraries and then pulling in the latest from source (v2.5.0) and reinstalling from that repository allowed the make process to get past all those previous errors.

Many thanks!

ZbychuButItWasTaken, 2022-09-17:
Well, I suppose that's fixed then. Ready to close?

K4KDR, 2022-09-17:
Yes - thank you!!

closed 2022-09-17 by ZbychuButItWasTaken