SatDump/SatDump #129: GEO Projection configurations and products missing

ZbychuButItWasTaken, 2022-09-17:
Yes they are gonna be added.

Aang23, 2022-09-17:
Todo : 
- [ ] GOES-16
- [ ] GOES-17
- [ ] ELEKTRO-L 2
- [ ] ELEKTRO-L 3
- [ ] Himawari-8
- [ ] GOES-13 (EWS-G1)

Aang23, 2022-09-17:
GOES-16 (Full Disk and CONUS), valid on GRB / HRIT
[image]
![test_goes_idk](https://user-images.githubusercontent.com/23421917/190859292-cf1c756c-568a-4f73-aa5b-6601477b377c.jpg

The overlay can be applied using `./apply_mapoverlay source_image.png settings.json output.png` build when -DBUILD_TOOLS=ON is set. It's only as products.cbor are not produced yet, and the tool I use to tune them.

[goes16_projs.zip](https://github.com/altillimity/SatDump/files/9591785/goes16_projs.zip)
)

Aang23, 2022-09-17:
ELEKTRO-L 3 (LRIT / HRIT)
I only had older data on hand from before it was officially operational. It's not perfectly aligned, might be due to that.
[image]
[elektrol3.zip](https://github.com/altillimity/SatDump/files/9591853/elektrol3.zip)

Aang23, 2022-09-17:
GK-2A (LRIT / HRIT)
[image]
[gk2a.zip](https://github.com/altillimity/SatDump/files/9591872/gk2a.zip)

Aang23, 2022-09-17:
GOES-17 (HRIT / GRB) FullDisk (no ""CONUS"" on Hand right now)
[image]
[goes17_fulldisk.zip](https://github.com/altillimity/SatDump/files/9591898/goes17_fulldisk.zip)

Aang23, 2024-04-23:
Closing this, as xRIT now generated projection configs. Some config samples to import other GEO images should be added on the website though.

closed 2024-04-23 by Aang23