SatDump/SatDump #130: [Win32] Crash in 'src-core/common/dsp/file_sink.cpp'

gvanem, 2022-09-17:
Simply starting `satdump-ui.exe` and exiting (Alt-F4) causes a crash in this destructor:
```c
    FileSinkBlock::~FileSinkBlock()
    {
        delete[] buffer_s8;
        delete[] buffer_s16;
    }
```

Call-stack:
```
  ntdll!RtlReportCriticalFailure+0x56
  ntdll!RtlpHeapHandleError+0x12
  ntdll!RtlpHpHeapHandleError+0x7a
  ntdll!RtlpLogHeapFailure+0x45
  ntdll!RtlpFreeHeapInternal+0x433
  ntdll!RtlFreeHeap+0x51
  satdump_core.dll!ucrtbase!_free_base+0x1b
  satdump_core!dsp::FileSinkBlock::~FileSinkBlock(void)+0x23
```

since `buffer_s8` and `buffer_s16` was allocated by `volk.dll`; potentially another CRT. 
So these has to be freed by `volk_free()`:
```diff
--- a/src-core/common/dsp/file_sink.cpp 2022-09-17 15:02:15
+++ b/src-core/common/dsp/file_sink.cpp  2022-09-17 17:06:06
@@ -11,8 +11,8 @@

     FileSinkBlock::~FileSinkBlock()
     {
-        delete[] buffer_s8;
-        delete[] buffer_s16;
+        volk_free(buffer_s8);
+        volk_free(buffer_s16);
     }

     void FileSinkBlock::work()
```

Is this really not an issue besides Windows?

*BTW*. I built Volk using an older version of MSVC. But I don't think that's related.

Aang23, 2022-09-17:
I can't reproduce that on Linux. Are you sure you only starting SatDump and doing nothing else?

ZbychuButItWasTaken, 2022-09-17:
Can confirm, doesn't happen on linux.

gvanem, 2022-09-17:
> Can confirm, doesn't happen on linux.

Well. That's because mixing CRTs is not an issue on Linux I guess.

gvanem, 2022-09-17:
> Are you sure you only starting SatDump and doing nothing else?

Yes.

gvanem, 2022-09-17:
And BTW, I see you call `volk_free()` elsewhere. Why not here too?

Aang23, 2022-09-17:
If I used volk_alloc, it should be used and that'd be an oversight!

Aang23, 2022-09-17:
@gvanem does your fix work? If it does, I'll gadly implement it.

gvanem, 2022-09-17:
> does your fix work? 

Sure. If I did not test that patch, I'd not wrote what I did,

closed 2022-09-17 by Aang23

gvanem, 2022-09-17:
BTW, this issue was about misusing memory from `_aligned_malloc()` and 
 `delete[]` (i.e. `free()`). There's no `delete_aligned[]` AFAIK. 
How this works on Linux, I've no idea. Perhaps everything is magically aligned to what Volk requires?

commit 343c20c4 (2022-09-17): Free with the correct Volk call. Fixes #130