SatDump/SatDump #131: Meteor LRPT image output?

K4KDR, 2022-09-17:
Caveat - I could EASILY be missing an option or doing something incorrectly...

Just tried the v1.0 GUI for the first time & recorded a **VHF Meteor-M2 LRPT** pass.

When I ran the 'offline processing' to create images, I got one for each channel but not the combo images that used to appear automatically.

Am I not seeing an option, does the software need to be run (or built) with some special option, or what?  

HUGE thanks!!

-Scott,  K4KDR

Aang23, 2022-09-17:
Well, now you either use the viewer to create them, or enable the "automatically process products" option in settings to do like it was before!

K4KDR, 2022-09-18:
Ahh - thanks SO much.

I'll have to explore what's available on that viewer tab.    And my bad -  I didn't understand what "automatically process products" meant.

This can be closed - thanks!!

closed 2022-09-18 by ZbychuButItWasTaken