SatDump/SatDump #132: FengYun-3 AHRPT Deframer out of sync

Foxiks, 2022-09-18:
[image]
The AHRPT decoder for the FY-3B has no sync for the deframer, while the Viterbi syncs.

Aang23, 2022-09-18:
Ok. I remember I forgot to fix that one... One of the tiny differences between FY's behavior on L and X-Band.

Will be fixed up. Didn't test FY-3B L-Band as much as it's not active anymore, and 3C is pretty challenging from Europe!

closed 2022-09-18 by Aang23

Foxiks, 2022-09-18:
Unfortunately this still doesn't work. Same problem with Metop

Aang23, 2022-09-18:
Uh what? Can you send the basebands? Didn't see that even once until now.

Foxiks, 2022-09-18:
Yes, give me 15-20 minutes to upload

Foxiks, 2022-09-18:
[image]
https://drive.google.com/file/d/1g0HiZPSOMrxUOgKx-N5og3Wk5yQsgUgT/view?usp=sharing MetOp

Aang23, 2022-09-18:
[image]

Aang23, 2022-09-18:
Oh wait! This may not be a bug in the decoders at all. It may be related to the recorder's scrolling behaviour perhaps.

Foxiks, 2022-09-18:
Hm this works a little weird. Look at this. https://youtu.be/tvQuw8sFX9w

Aang23, 2022-09-18:
Yep. That's what I am talking about actually! Will wait to check with FY and debug.

Foxiks, 2022-09-18:
the same with FY. Deframer for FY works the same as with metop

Aang23, 2022-09-18:
I see. I think I know why. It's a bug that can only occur when replaying a baseband and not when on an actual signal.
I'll wait for the FY baseband to check still, but otherwise I'm already thinking of a few ways to fix that properly.

Foxiks, 2022-09-18:
Also, sometimes during offline decoding from IQ recording, the deframer shows the "syncing" status and this is very strongly reflected in the photo. The previous version of satdump does not have this problem.

Aang23, 2022-09-18:
I've gotten a fix implemented for MetOp (again, it's a behaviour that only happens because the demodulator is suddenly fed a strong signal, strong enough for the Viterbi to lock on the wrong state). I'm gonna test on FY to confirm everything works before I push though.

Aang23, 2022-09-18:
Just pushed some fixes. Should have helped.

Foxiks, 2022-09-18:
Excellent! Thank you!

Foxiks, 2022-09-19:
[image]
Sometimes, after a momentary loss of signal, the deframer goes into the "syncing" status. It does not depend on signal strength

Aang23, 2022-09-19:
Do you have a baseband sample of that happening so I can test here? Could not see that with any of my basebands.

Foxiks, 2022-09-19:
Yes. But I need to upload this... Give me some time)

Aang23, 2022-09-19:
No worries! I will wait! Thanks

Foxiks, 2022-09-19:
Unfortunately, my entry has a weight of more than 26 GB. I won't be able to upload it today and will upload it tomorrow

Aang23, 2022-09-19:
No problem!
You can use https://www.swisstransfer.com if necessary.

Foxiks, 2022-09-20:
unfortunately i have a very low speed for [swisstransfer](https://www.swisstransfer.com/). I will upload it throughout tomorrow

Aang23, 2022-09-20:
No worries!

Foxiks, 2022-09-21:
https://drive.google.com/file/d/1odzinGOBnT1_cCbYweuNT1vwjcl_vTGf/view?usp=sharing

Aang23, 2022-09-21:
Hmmm interestingly enough, here it did *not* unlock even once (apart from the few sampledrops here and there). I will test on Windows, maybe the issue is specific there.

Aang23, 2022-09-21:
Just to clarify : Whenever you see a spike like this it due to sampledrop in the baseband : so because your SSD/HDD was not fast enough to save what you were recording. Nothing I can do fix that unfortunately!

[image]
[image]

Otherwise, everything seems mostly fine. Still re-tuning some things and gonna do a bit more testing though.

Foxiks, 2022-09-21:
Hm, okay. But for me, after such a "burst", sometimes the deframer does not receive normal data and goes into the "SYNCING" status

Aang23, 2022-09-21:
It also does here (which is expected), but does stay stuck in SYNCING state. I will try on Windows soon.

Foxiks, 2022-09-21:
okay

commit 99bd8f2b (2022-09-18): Fix automatic IQ swap for FY 3/4. Fixes #132