SatDump/SatDump #134: Icon entry in .desktop file lacks file extension

shuyuan-liu, 2022-09-19:
The `Icon` entry in `satdump.desktop` lacks the `.png` file extension and it causes GNOME 42.4 to not show any icon for the program:

[image]

The [freedesktop.org specification for desktop entries](https://specifications.freedesktop.org/desktop-entry-spec/desktop-entry-spec-latest.html#entries) states that "if the name is an absolute path, the given file will be used", which probably means the file extension needs to be included. Once I added `.png` to the path, the icon showed fine.

shuyuan-liu, 2022-09-19:
Curiously, the `.png` was there before but was removed in d9b8801908ab8009201651e0c5d513cea31ff501. Was it done for any particular reason?

Aang23, 2022-09-19:
I'd guess back then I read it should not be provided and removed it. Thanks for reporting!

Do you want to submit a PR or should I fix it?

closed 2022-09-19 by Aang23

shuyuan-liu, 2022-09-19:
Sorry I didn't expect such a fast reply — and thanks for the fix!

Aang23, 2022-09-19:
No worries! Thanks!

commit d1ddef94 (2022-09-19): Add .png. Fixes #134