SatDump/SatDump #135: Feature idea for NOAA

kord1e, 2022-09-21:
AFAIK there's more in the DSB signal than just HIRS data... (see @nebarnix/project-desert-tortoise)
would be cool to have Satdump unpack this data as well!

Aang23, 2022-09-21:
There is a bit more yes, but not so much more. HIRS is the main payload.

It does contain SEM though, which would indeed be a nice addition alongside perhaps DCS and S/C TLM. Though instrumentation I'd say is the main focus.

DCS Formats are extremely undocumented unfortunately so even though I'd love supporting those on all sats doing so, hasn't been an easy thing.

ZbychuButItWasTaken, 2022-09-22:
SEM is definitely planned, as well as SBUV/2.

ZbychuButItWasTaken, 2022-10-18:
SEM was added in 6e8ed03e859960238dd12f0264ed6dee9edb1107. For now other things in TIP will not be worked on and here is why:
- SEM and HIRS are the most interesting instruments in TIP and they are already supported
- DCP is extremly unducumented so that's a no-go
- SBUV would be cool but for now I will not be implementing it due to poor documentation and comlex format (maybe will work on it eventually, but not anytime soon most likely)
- telmetry data sent in TIP frames without documentation is completly useles thus extracting it is pointless
- no point in extracing instrument housekeeping data either, only parts of it are needed for calibration and that will be haneld internally

So really everything that is of any value is already extracted (except SBUV, but the amount of work is worth more than the data from it at this stage) and everything else is not of any use for processing the data.

Not closing for now though as SBUV is still in consideration.

ZbychuButItWasTaken, 2023-01-23:
We've decided no support for SBUV/2 will be added

closed 2023-01-23 by ZbychuButItWasTaken